project-chip/connectedhomeip #26815: Java controller compilation becomes flaky on fromJsonString

yunhanw-google, 2023-05-24:
INFO    warning: flag is not supported by this version of the compiler: -Xlint:all
INFO    warning: flag is not supported by this version of the compiler: -Xlint:deprecation
INFO    /__w/connectedhomeip/connectedhomeip/examples/java-matter-controller/third_party/connectedhomeip/src/controller/java/tests/chip/jsontlv/JsonToTlvToJsonTest.kt:28:21: error: unresolved reference: fromJsonString
INFO    import chip.jsontlv.fromJsonString
INFO                        ^
INFO    /__w/connectedhomeip/connectedhomeip/examples/java-matter-controller/third_party/connectedhomeip/src/controller/java/tests/chip/jsontlv/JsonToTlvToJsonTest.kt:29:21: error: unresolved reference: toJsonString
INFO    import chip.jsontlv.toJsonString
INFO                        ^
INFO    /__w/connectedhomeip/connectedhomeip/examples/java-matter-controller/third_party/connectedhomeip/src/controller/java/tests/chip/jsontlv/JsonToTlvToJsonTest.kt:42:28: error: unresolved reference: fromJsonString
INFO        assertThat(TlvWriter().fromJsonString(jsonOriginal)).isEqualTo(tlvEncoding)
INFO                               ^
INFO    /__w/connectedhomeip/connectedhomeip/examples/java-matter-controller/third_party/connectedhomeip/src/controller/java/tests/chip/jsontlv/JsonToTlvToJsonTest.kt:43:39: error: unresolved reference: toJsonString
INFO        assertThat(TlvReader(tlvEncoding).toJsonString())
INFO                                          ^
INFO    /__w/connectedhomeip/connectedhomeip/examples/java-matter-controller/third_party/connectedhomeip/src/controller/java/tests/chip/jsontlv/JsonToTlvToJsonTest.kt:46:30: error: unresolved reference: fromJsonString
INFO          assertThat(TlvWriter().fromJsonString(jsonExpected)).isEqualTo(tlvEncoding)
INFO                                 ^
INFO    /__w/connectedhomeip/connectedhomeip/examples/java-matter-controller/third_party/connectedhomeip/src/controller/java/tests/chip/jsontlv/JsonToTlvToJsonTest.kt:162:36: error: unresolved reference: toJsonString
INFO        assertThat(TlvReader(encoding).toJsonString())
INFO                                       ^
INFO    /__w/connectedhomeip/connectedhomeip/examples/java-matter-controller/third_party/connectedhomeip/src/controller/java/tests/chip/jsontlv/JsonToTlvToJsonTest.kt:388:69: error: unresolved reference: toJsonString
INFO        assertFailsWith<IllegalArgumentException> { TlvReader(encoding).toJsonString() }

bzbarsky-apple, 2023-05-24:
@yunhanw-google Note https://github.com/project-chip/connectedhomeip/pull/26781

joonhaengHeo, 2023-05-25:
@yunhanw-google @yufengwangca 
I also checked for a similar Build Error.
The exact reason is unknown, but it is confirmed that the same class is included in multiple jar files that are created.

In order to remove this, some code positions are changed and processed, so no error occurs.
(This is considered necessary for file separation according to the purpose of each jar file.)

I made the commit below as a sample, please review if there is a better way.

https://github.com/joonhaengHeo/connectedhomeip/commit/2dc8b08d3d3f3649b1ebbe0a8a19b2164e821271

yufengwangca, 2023-05-25:
+ @emargolis 

It looks we have cyclical include after we refactoring jsontotlv from tlv kotlin library, I have a workaround https://github.com/project-chip/connectedhomeip/pull/26781, but I agree we should reorg the files to remove these cyclic dependency.

PR #26893 (2023-05-27): Make output class directory unique for java
This looks to fix the `chip.jsontlv.toJsonString` error that we seem to be getting (i.e. fixes #26815)

From what I can tell, each compilation used to codegen under `classes` subdirectory, so if different classes compile things, we may end up with different versions of the same code.

From my local tests, applying this change fixed an always-reproducible fromJsonString error, so I assume this works on current codebase as well.

Before this change, every jar file built ended up containing *ALL* the class files that were built for all targets (but rand [... cut]
closed 2023-05-27 by andy31415