project-chip/connectedhomeip #26821: [BUG] Check endpoint index with dynamic endpoint

joe-kohler, 2023-05-25:
### Reproduction steps

read/write onoff server dynamic endpoint
in function EmberEventControl * OnOffServer::getEventControl(EndpointId endpoint)
it calls emberAfGetClusterServerEndpointIndex.
In file attribute-storage.cpp line 883, "    VerifyOrDie(fixedClusterServerEndpointCount <= FIXED_ENDPOINT_COUNT);
this should be "VerifyOrDie(fixedClusterServerEndpointCount <= FIXED_ENDPOINT_COUNT + CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT);"

My project with dynamic endpoint on/off crash when call this.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

5d459ceb

### Platform

esp32

### Platform Version(s)

4.4

bzbarsky-apple, 2023-05-25:
@joe-kohler No, the check that's there is correct.  fixedClusterServerEndpointCount is supposed to be the number of _fixed_ endpoints that have the relevant cluster on them.  So it must be no larger than the total number of fixed endpoints.

In your project, what are the values of `EMBER_AF_ON_OFF_CLUSTER_SERVER_ENDPOINT_COUNT` and `FIXED_ENDPOINT_COUNT`?  And if the former is bigger than the latter, where is that value of `EMBER_AF_ON_OFF_CLUSTER_SERVER_ENDPOINT_COUNT` coming from in your case?

joe-kohler, 2023-05-26:
In my project FIXED_ENDPOINT_COUNT is 1, the root device.
#define EMBER_AF_ON_OFF_CLUSTER_SERVER_ENDPOINT_COUNT (2)
Two eps for two control points.
The OnOff endpoint is create dynamic and add to matter system, just as most bridge does.

joe-kohler, 2023-05-26:
Thanks bzbarsky,
I changed EMBER_AF_ON_OFF_CLUSTER_SERVER_ENDPOINT_COUNT to 0, this definition is for fixed OnOff ep.

joe-kohler, 2023-05-30:
In OnOff server, the endpoint count is like:
static constexpr size_t kOnOffMaxEnpointCount =
    EMBER_AF_ON_OFF_CLUSTER_SERVER_ENDPOINT_COUNT + CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT;
static EmberEventControl gEventControls[kOnOffMaxEnpointCount];

In **ColorServer**, the endpoint count is count without dynamic endpoint count, like

EmberEventControl eventControls[EMBER_AF_COLOR_CONTROL_CLUSTER_SERVER_ENDPOINT_COUNT];
src/app/clusters/color-control-server/color-control-server.h line 255

bzbarsky-apple, 2023-05-30:
> In ColorServer, the endpoint count is count without dynamic endpoint count, like

Yes, the color control server is broken here. That's not a problem with the check this issue was about.  Filed https://github.com/project-chip/connectedhomeip/issues/26959

closed 2023-05-30 by bzbarsky-apple