project-chip/connectedhomeip #26841: [Feature] IDM cluster device tests YAML support

manjunath-grl, 2023-05-25:
### Feature description

Summary: 
Need IDM cluster YAML support for the below mentioned steps or sample script if implemented already.
Effected tests:
TC-IDM-1.2
TC-IDM-2.2
TC-IDM-3.2
TC-IDM-4.2
TC-IDM-4.3
TC-IDM-4.4
TC-IDM-5.2
TC-IDM-6.1
TC-IDM-6.2
TC-IDM-7.1
TC-IDM-8.1
Feature required: 
1. Need YAML support to send the [ `Invoke Request Message` , `Read Request Message`, `WriteRequestMessage`, `subscription      request message`, `Timed Request message` ] from `TH `to `DUT` with single or multiple `PATHS` and `CommandDataIB` -->' which has the specific Endpoint, Specific Cluster and Specific Command.'
2. Setup TH2 such that it does `not have access` or `have access` to an attribute on a specific cluster and endpoint.
3. TH sends `Read Request Message` to DUT with `EventRequests` set to a specific `event` from a specific `cluster` on a specific `endpoint` on a specific `node`
4. TH sends the `Invoke Request Message `to the DUT with a valid `CommandDataIB` and `SuppressResponse` set to True
5. Need YAML support for all steps in IDM-7.1 & 8.1 testcases

### Platform

other

### Anything else?

IDM-1.2
[image]
[image]
IDM-2.2
[image]
IDM-3.2
[image]
IDM-4.2
[image]
IDM-5.2
[image]
IDM-6.4
[image]

manjunath-grl, 2023-05-25:
Breakdown for issue: https://github.com/project-chip/connectedhomeip/issues/23232

bzbarsky-apple, 2023-05-25:
> Need YAML support to send the [ Invoke Request Message , Read Request Message, WriteRequestMessage, subscription      request message, Timed Request message ] from TH to DUT with single or multiple [...]

This is duplicating things from https://github.com/project-chip/connectedhomeip/issues/23232, no?

>  Setup TH2 such that it does not have access or have access to an attribute on a specific cluster and endpoint.

OK, so set ACLs.  What's missing in YAML here?

>    TH sends Read Request Message to DUT with EventRequests set to a specific event from a specific cluster on a specific endpoint on a specific node

readEvent exists, no?

> Breakdown for issue: https://github.com/project-chip/connectedhomeip/issues/23232

Honestly, a breakdown by _test_ is not useful @manjunath-grl . What's useful is a breadkdown per-missing-piece-of-functionality, one issue per missing piece of functionality.  And the issues involved should clearly describe exactly what's missing.

manjunath-grl, 2023-05-26:
@bzbarsky-apple I will close this issue. Let me go through this and open a dedicated issue for the functionality.

closed 2023-05-26 by manjunath-grl