project-chip/connectedhomeip #26858: [Question] Some questions about multi-admins/fabrics

IcyWaterDL, 2023-05-26:
1. Can I add multiple admins to one node in the same fabric?
2. Help me differentiate between multi-admins and multi-fabrics, please.

bzbarsky-apple, 2023-05-26:
> Can I add multiple admins to one node in the same fabric?

Yes.  Just set up the ACLs on the node accordingly.

> Help me differentiate between multi-admins and multi-fabrics, please.

A "fabric" is a collection of nodes that can talk to each other, because they have a shared root of trust and can use the CASE protocol to set up an encrypted connection.  A node can be in multiple fabrics by having multiple operational certificates.

An "admin" is a node that has been granted administrative privileges on some other node, via the ACLs set there.

You can have multiple admins all on the same fabric, if you set up your ACLs that way.  Or you could remove all the admin ACLs and have no admins at all, though that's generally a bad idea.

closed 2023-05-26 by bzbarsky-apple

IcyWaterDL, 2023-05-29:
can i add other matter controller into the same fabric?

bzbarsky-apple, 2023-05-30:
> can i add other matter controller into the same fabric?

Yes, as long as all the controllers involved have operational certificates chaining up to the same root.

IcyWaterDL, 2023-05-31:
> > can i add other matter controller into the same fabric?
> 
> Yes, as long as all the controllers involved have operational certificates chaining up to the same root.

Can you give me some instructions on how to deploy it?