project-chip/connectedhomeip #26865: [Documentation] How can I get response data with chip-tool command?

lancersky2017, 2023-05-26:
### Documentation issues

Hi, Is there a guidance document on how to use chip-tool command or API to get response data in a readable format?

bzbarsky-apple, 2023-05-26:
It depends on what you mean by "readable".

If you just want the command payloads, see the https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/chip_tool_guide.md#using-message-tracing

If you want to see something else... chances are chip-tool is not set up for it.  It will log things like command responses and attribute read/subscribe results, for cases when it knows the schema to parse as, and it can log incoming TLV, but that's it.

So what _exactly_ are you trying to do?

lancersky2017, 2023-05-30:
Hi @bzbarsky-apple ,
I want to parser the chip-tool response data to JSON and send it to the Cloud. If the data is in the log, it is useless for our actual application. Is there any other way?

bzbarsky-apple, 2023-05-30:
@lancersky2017 I believe `chip-tool interactive server` lets you start a chip-tool that you can then connect to via websocket to send it JSON commands and get JSON responses.  I don't know that this is documented particularly well, though.  @vivien-apple would know for sure.

lancersky2017, 2023-07-05:
Need to update to the latest version.

closed 2023-07-05 by lancersky2017

nmtoan91, 2023-12-18:
@vivien-apple @bzbarsky-apple 
My Websocket client doesn't receive messages for subscribe command.
Is it true that the subscription messages are sent viva WebSocket?

bzbarsky-apple, 2023-12-19:
@nmtoan91 I dont know.