project-chip/connectedhomeip #26868: Make android  app act as an OTA provider

MrLiuYunPing, 2023-05-26:
### Feature description

...
I have done this in iOS, because the Matter.framework has supported to set an OTA delegate, and has some test cases related. But, in android,  I announce the controller nodeId to the device to update, when I set OTA delegate using endpoint 0, it always failed(emberAfFindClusterServerEndpointIndex method returns Invalid). When i look the logs on the device, it shows: Received Command Response Status for Endpoint=0 Cluster=0x0000_0029 Command=0x0000_0000 Status=0x7f . . .Status=0x7f whichs means unsupported endpoint. I don't know why iOS success but android failed?

### Platform

android

bzbarsky-apple, 2023-05-26:
> I don't know why iOS success but android failed?

iOS implements `emberAfFindClusterServerEndpointIndex` specifically so OTA can work.  See https://github.com/project-chip/connectedhomeip/blob/19ae369ee57b828b3934bea599f1e2d000d32730/src/darwin/Framework/CHIP/MTRIMDispatch.mm#L230-L240

Android might need to do something similar.

srdantas, 2023-05-26:
how could you set ota provider on android @MrLiuYunPing ? may you share your code here? I never found documentarion for support this feature

swap9391, 2023-06-05:
Any update what steps need to perform on Android code achieve OTA in android?

yusheng-jia, 2023-06-26:
@bzbarsky-apple Hi Could you give me a simple example how to make OTA process on iOS side? I didn't find any docs on this, I want to do OTA function on iOS, Thank you .

yusheng-jia, 2023-06-26:
@MrLiuYunPing Hi could you tell me how you implemented on iOS? Thank you

bzbarsky-apple, 2023-06-26:
@yusheng-jia On iOS, set `otaProviderDelegate` in the `MTRDeviceControllerFactoryParams` to a non-nil value before starting up the controller factory.  Then you will get calls to your delegate implementation as needed.

yusheng-jia, 2023-06-27:
> @yusheng-jia On iOS, set `otaProviderDelegate` in the `MTRDeviceControllerFactoryParams` to a non-nil value before starting up the controller factory. Then you will get calls to your delegate implementation as needed.

Thanks, I will do that.

yunhanw-google, 2023-07-27:
I will take a look today. thanks

yunhanw-google, 2023-07-28:
@MrLiuYunPing  @swap9391 @yusheng-jia https://github.com/srdantas)may I know your requirement on android for ota-provider? for example, you wanna use android-chip-tool or your android app to announce there is ota software, then send this image to your wifi-device using bdx over wifi? I think currently there is no java/jni api for setDelegate for ota-provider, how do you do that, wanna get more details so that we can plan the feature. thanks

yusheng-jia, 2023-07-28:
> @MrLiuYunPing @swap9391 @yusheng-jia https://github.com/srdantas)may I know your requirement on android for ota-provider? for example, you wanna use android-chip-tool or your android app to announce [...]

Thank you for support, you're totally right . I did this on iOS, iOS has MTROTAProviderDelegate protocol, it has very detailed method to implement the whole OTA process on App side.

yunhanw-google, 2023-07-28:
@yusheng-jia are you in matter slack channel, if yes, what is your slack name?

yusheng-jia, 2023-08-01:
> 
Sorry for miss this message, I'm not in slack channel.

eling22, 2023-08-21:
@yunhanw-google Hi, I am also surveying this feature. Is an OTA provider currently being developed?

yunhanw-google, 2023-09-07:
@eling22 are you in slack, could you find me with yunhanw in slack? Wanna also understand your requirement, thanks

yunhanw-google, 2023-09-15:
@eling22 and @MrLiuYunPing we are actively supporting this feature, we would create several PR for this work soon.

Shahmeera, 2023-11-06:
I don't have knowledge of an android app, but I was facing same issue with linux PC. In your android app if you want to be as an OTA provider then OTA provider cluster must be in server mode. I have resolved this error by doing following changes.
(1) updated matter version 1.2 from 1.1 because updated ZAP version can support **OTA Provider cluster as a Server**
(2) Enabled OTA provider cluster as a server.
By doing these, you will be able to resolve this.

yunhanw-google, 2023-11-06:
@Shahmeera @eling22 @MrLiuYunPing @yusheng-jia @swap9391 @srdantas  we have officially added the OTA support in android side, please check-in, thanks
https://github.com/project-chip/connectedhomeip/pull/30145

closed 2023-11-06 by yunhanw-google