project-chip/connectedhomeip #26881: [Documentation] [Silabs] Broken links since PR #26766 Rename examples folder

lboue, 2023-05-26:
### Documentation issues

[Documentation] [Silabs] Broken links since PR #26766 Rename examples folder

## Build
```
cd ~/connectedhomeip
  ./scripts/examples/gn_silabs_example.sh ./examples/lighting-app/silabs/efr32/ ./out/lighting-app BRD4187C
+ PYTHON_PATH=/home/user/connectedhomeip/.environment/pigweed-venv/bin/python3
+ BUILD_DIR=./out/lighting-app/BRD4187C
+ echo BUILD_DIR=./out/lighting-app/BRD4187C
BUILD_DIR=./out/lighting-app/BRD4187C
+ '[' '' == true ']'
+ '[' false == true ']'
+ DOTFILE=./examples/lighting-app/silabs/efr32//openthread.gn
+ '[' false == true ']'
+ gn gen --check --script-executable=/home/user/connectedhomeip/.environment/pigweed-venv/bin/python3 --fail-on-unused-args --export-compile-commands --root=./examples/lighting-app/silabs/efr32/ --dotfile=./examples/lighting-app/silabs/efr32//openthread.gn '--args=silabs_board="BRD4187C" sl_matter_version_str="v1.1-master-1aca004d68-dirty" ' ./out/lighting-app/BRD4187C
ERROR Root source path not found.
The path "./examples/lighting-app/silabs/efr32/" doesn't exist.
user@matter-vm:~/connectedhomeip$

```

### Platform

efr32

bzbarsky-apple, 2023-05-26:
@jepenven-silabs

lboue, 2023-11-02:
Did you fix it?

lboue, 2024-08-28:
Fixed

closed 2024-08-28 by lboue