project-chip/connectedhomeip #26894: TLV decoding should validate UTF-8 data

bzbarsky-apple, 2023-05-26:
When we try to decode a "UTF8 string" from TLV, the decoder should (optionally?) verify that the string is in fact UTF-8.

Need to figure out which validations this should do, specifically.

tcarmelveilleux, 2023-05-26:
The default should be to replace bad glyphs with Unicode U+FFFD (REPLACEMENT CHARACTER) https://www.fileformat.info/info/unicode/char/fffd/index.htm which has the obvious meaning, when seen, that trash was present. It leaves most of the information as cognate.

andy31415, 2023-05-26:
The thing to recognize is also we have matter certified devices that may actually do this, so being lenient seems better than failing a whole read/subscribe.

bzbarsky-apple, 2023-05-26:
You would fail a single attribute, I would hope, not the whole thing.

bzbarsky-apple, 2023-05-26:
But again, the important part is clearly defining what we consider "valid" here.

tcarmelveilleux, 2023-05-27:
> You would fail a single attribute, I would hope, not the whole thing.

Actually, on Android cluster handling generated code until April 22, 2023, it was an exception, likely uncaught due to the path, and between April 22, 2023 and May 26, 2023, this was a crash (abort) in JNI code (uncatchable as it breaks internal consistency of JVM).

So much worse than a single attribute fail: seeing the attribute could cause a crash depending on app/component. I would wager many clients did not properly catch the exceptions and "do something sane", since it occurred way deep in decoding.

bzbarsky-apple, 2023-05-27:
Yes, I know the Android code ended up with a fatal error here.  But that's not an immutable property of either the Android code or other code.