project-chip/connectedhomeip #26905: [Documentation] WindowCOnvering Cluster Spc vs XML inconsistencies

Apollon77, 2023-05-26:
### Documentation issues

* The Attributes 0x10 till 0x13 have a max of 0xffff in the XML and a max of 65534 aka 0xfffe in the Matter Specs 1.1. What is correct? Same for the default value on 0x11 and 0x13

### Platform

core (please add to version below)

bzbarsky-apple, 2023-05-26:
@jmeg-sfy please take a look?

jmeg-sfy, 2023-05-30:
We should correct the xml following the specs
Specs on these attributes is legacy ZigBee not idea why those are uint16 maximum minus 1 TBH 
Theses are not NULLABLE

jmeg-sfy, 2023-05-30:
ABS is not used in Matter today

bzbarsky-apple, 2023-05-30:
>  Specs on these attributes is legacy ZigBee got no idea why those are uint16 maximum minus 1 TBH

Is it possible this is just a transcription error from ZigBee, and we should fix it in the spec @jmeg-sfy ?

Apollon77, 2023-06-01:
> ABS is not used in Matter today

Does that mean that this feature should be "!Matter" because zigbee only or is it intended to be used somewhen later also for matter?

Also for number Range doesn't Nesbit is full uInt16?

And all Nullable are wrong or which are ok?

Apollon77, 2023-06-02:
@jmeg-sfy hey, just as backgrounder for my questions. I work on matter.js and we are just about to implement all clusters there and so I stumbled over the inconsistencies. Would be cool to get some clarity, else I need to guess what's right ;-) (and most likely you are more expert then me in window coverings ...)

jmeg-sfy, 2023-06-12:
> > ABS is not used in Matter today
> 
> Does that mean that this feature should be "!Matter" because zigbee only or is it intended to be used somewhen later also for matter?
> 
> Also for number Range seems should be full uInt16?
> 
> And all Nullable are wrong or which ones are ok?

The ranges for the uint16 should be in full IMO

And we should have the nullable at full ranges as well

bzbarsky-apple, 2023-06-12:
> The ranges for the uint16 should be in full IMO

@jmeg-sfy spec issue needed?

Apollon77, 2026-10-22:
Attributes 0x0010-0x0013 are now Zigbee-only (Z) in the spec, and the XML max matches the spec. Closing.

closed 2026-10-22 by Apollon77