project-chip/connectedhomeip #26912: Python-repl error

aravindone, 2023-05-28:
### Reproduction steps

whenever i used to python commands involving cluster keyword it throws up the error

In [39]: matterhelp(Clusters.UnitTesting.Commands.TestAddArguments)
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
Cell In[39], line 1
----> 1 matterhelp(Clusters.UnitTesting.Commands.TestAddArguments)

NameError: name 'Clusters' is not defined

@bzbarsky-apple  please comment

### Bug prevalence

every time calling cluster 

### GitHub hash of the SDK that was being used

bfa05804ac3c224c9e70921a857ffcac32ccc7f4

### Platform

python

bzbarsky-apple, 2023-05-28:
@aravindone you might want to ask someone who knows something about the Python repl, not me....  That said, I am not sure who, if anyone, is maintaining it now.  Maybe @tehampson ?

tehampson, 2023-05-29:
@aravindone , What happens if you do `import chip.clusters as Clusters`?

aravindone, 2023-05-29:
problem resolves @tehampson  thanks for the comment. It works. i will close this ticket

closed 2023-05-29 by aravindone