project-chip/connectedhomeip #26915: [BUG] Camera doesn't display preview frame in Google Play Service when use GPS to do matter provisioning.

M201370367, 2023-05-29:
### Reproduction steps

GPS version: 23.17.60
phone : pixel 4a Android 13

Reproduction steps:
1, Open GHSAFM demo and press "+" button.
GHSAFM demo link: https://github.com/google-home/sample-app-for-matter-android
2, The Matter Scan page of Google Play Service will be shown. But the screen won't show any frames. The app has camera permission.
log:
[matter_commission_error_scan.log](https://github.com/project-chip/connectedhomeip/files/11587464/matter_commission_error_scan.log)

05-25 16:14:27.248 20330 20330 E AndroidRuntime: FATAL EXCEPTION: main
05-25 16:14:27.248 20330 20330 E AndroidRuntime: Process: com.google.android.gms.ui, PID: 20330
05-25 16:14:27.248 20330 20330 E AndroidRuntime: java.lang.IllegalArgumentException: No supported surface combination is found for camera device - Id : 0.  May be attempting to bind too many use cases. Existing surfaces: [AttachedSurfaceInfo{surfaceConfig=SurfaceConfig{configType=PRIV, configSize=PREVIEW}, imageFormat=34, size=720x480, dynamicRange=DynamicRange@fed9bcc{format=FORMAT_SDR, bitDepth=8}, targetFrameRate=null}, AttachedSurfaceInfo{surfaceConfig=SurfaceConfig{configType=YUV, configSize=VGA}, imageFormat=35, size=640x480, dynamicRange=DynamicRange@fed9bcc{format=FORMAT_SDR, bitDepth=8}, targetFrameRate=null}] New configs: [m.ano@982e5e2, m.amr@d92ed73]
05-25 16:14:27.248 20330 20330 E AndroidRuntime: 	at m.eef.d(:com.google.android.gms.optional_home@231760065@23.17.60 (100400-0):68)
05-25 16:14:27.248 20330 20330 E AndroidRuntime: 	at m.eex.o(:com.google.android.gms.optional_home@231760065@23.17.60 (100400-0):2)

Screenshot:
[image]

### Bug prevalence

Anytime.

### GitHub hash of the SDK that was being used

GPS version: 23.17.60

### Platform

android

### Platform Version(s)

phone : pixel 4a Android 13

andy31415, 2023-05-30:
This does not seem the right place for issues regarding google-home sample app issues.

Please add it to https://github.com/google-home/sample-app-for-matter-android/issues instead.

closed 2023-05-30 by andy31415

andy31415, 2023-05-30:
I see the corresponding bug is https://github.com/google-home/sample-app-for-matter-android/issues/134

jonsmirl, 2023-05-30:
This is a problem in GPS...
Reboot made no difference.
~~~
05-30 10:19:06.253 18425 18425 I SurfaceView@a1cd525: Changes: creating=false format=false size=false visible=false alpha=false hint=false mUseAlpha=false visible=false left=false top=true
05-30 10:19:06.253 18425 18425 I SurfaceView@a1cd525: 169661733 Cur surface: Surface(name=null)/@0xb05af20
05-30 10:19:06.254 18425 18425 I SurfaceView@a1cd525: applyTransactionOnVriDraw: vri = ViewRootImpl@732c2fd[SetupDeviceActivity] fRS = false t = 0xb4000075c70a3000 android.view.SurfaceView.performSurfaceTransaction:959 android.view.SurfaceView.updateSurface:1105 android.view.SurfaceView.lambda$new$0$android-view-SurfaceView:200 
05-30 10:19:06.254 18425 18489 W CaptureSession: Opening session with fail OPENING
05-30 10:19:06.254 18425 18489 W CaptureSession: java.lang.NullPointerException
05-30 10:19:06.254 18425 18489 W CaptureSession: 	at m.bgm.g(:com.google.android.gms.optional_home@231712065@23.17.12 (100400-0):0)
05-30 10:19:06.254 18425 18489 W CaptureSession: 	at m.xz.c(:com.google.android.gms.optional_home@231712065@23.17.12 (100400-0):25)
05-30 10:19:06.254 18425 18489 W CaptureSession: 	at m.xs.a(:com.google.android.gms.optional_home@231712065@23.17.12 (100400-0):0)
05-30 10:19:06.254 18425 18489 W CaptureSession: 	at m.aql.run(:com.google.android.gms.optional_home@231712065@23.17.12 (100400-0):5)
05-30 10:19:06.254 18425 18489 W CaptureSession: 	at m.aqh.run(:com.google.android.gms.optional_home@231712065@23.17.12 (100400-0):5)
05-30 10:19:06.254 18425 18489 W CaptureSession: 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
05-30 10:19:06.254 18425 18489 W CaptureSession: 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
05-30 10:19:06.254 18425 18489 W CaptureSession: 	at java.lang.Thread.run(Thread.java:1012)

~~~