project-chip/connectedhomeip #26916: [Question] Some questions about multi-admins

IcyWaterDL, 2023-05-29:
**Can I add another Matter Controller to the same fabric?** This means that I want to have two or more admins with access to the same fabric.
**_I want to undertake a project like this:_** I have a mobile application running a paired Matter Controller controlling a Matter light bulb.
=> I want to add another phone (running matter-controller, within the same ecosystem) to the same fabric to control that light bulb.

bzbarsky-apple, 2023-05-30:
@IcyWaterDL How is this different from https://github.com/project-chip/connectedhomeip/issues/26858?

IcyWaterDL, 2023-05-31:
> @IcyWaterDL How is this different from #26858?

Can you give me some instructions on how to deploy it?

bzbarsky-apple, 2023-05-31:
@IcyWaterDL how do deploy what?  Your certificate infrastructure?  Something else?

IcyWaterDL, 2023-05-31:
> @IcyWaterDL how do deploy what? Your certificate infrastructure? Something else?
I apologize for any confusion. I want to add another Matter Controller to the same fabric. Could you please provide me with the necessary requirements (such as certificates, etc.) and a concise set of implementation steps? Your assistance would be greatly appreciated.

bzbarsky-apple, 2023-05-31:
@IcyWaterDL The necessary requirement for just joining a fabric is that the new controller have the right root certificate (matching the other things on the fabric), the right intermediate certificate if your CA is using intermediates, and an operational certificate that chains up to the root in question.  It will need the fabric id, though it can extract that from the operational certificate if necessary.

The details of how you initialize the controller with those certificates depends on exactly how you are creating/initializing your controllers to start with.

The details of how you get your hands on the relevant certificates depend very very strongly on how your fabric is getting  its certificates issued to start with.

To have the new controller be able to control devices, it will also need to:

1) Discover the devices somehow.  How that's accomplished depends on what exactly you are trying to do (e.g. interact with one specific device, interact with all).
2) Have ACLs on the devices that will allow it to interact with them.  These ACLs will need to be set up by the existing controller, which presumably has administrator privileges on all the devices.  There are various ways of doing this depending on your exact goals, but if you plan to just have two controllers and want them both to be admins the simplest thing is to just list both controllers' node ids in the subject fields of all the ACLs involved.