project-chip/connectedhomeip #26917: [BUG] Log out in ESP32

joe-kohler, 2023-05-29:
### Reproduction steps

When log out in ESP32 platform like 
 ChipLogProgress(AppServer, "xxxxx");

the Logout level in SDKconfig is Info
CONFIG_LOG_DEFAULT_LEVEL_INFO=y
CONFIG_LOG_DEFAULT_LEVEL=3

the matter sdk neabled appserver logout as
#define CHIP_CONFIG_LOG_MODULE_AppServer 1

sometimes the logout appserver without content like
I (770) chip[SVR]: 

### Bug prevalence

always

### GitHub hash of the SDK that was being used

cfb9994b

### Platform

esp32

### Platform Version(s)

4.4

### Anything else?

In function void ENFORCE_FORMAT(3, 0) LogV(const char * module, uint8_t category, const char * msg, va_list v)
line 45             
      esp_log_writev(ESP_LOG_INFO, tag, msg, v);
should be refine as
       esp_log_writev(ESP_LOG_INFO, module, msg, v);

For the sdk compare the module name with const module name memory address, not the content. the tag as a local variable, the memory changed time by time.

dhrishi, 2023-05-29:
@shubhamdp Please check

bzbarsky-apple, 2023-05-30:
> For the sdk compare the module name with const module name memory address, not the content

That seems a bit odd.  Where is that happening?

dhrishi, 2024-10-25:
@shubhamdp