project-chip/connectedhomeip #26921: [BUG]  to write command not working

aravindone, 2023-05-29:
### Reproduction steps

Hello , 
im trying to write value to cluster but the cluster is not supporting i dono why im getting this output ,

I tried other clusters 

https://raw.githubusercontent.com/aravindone/aravindone/main/write%20error

### Bug prevalence

always

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/commit/bfa05804ac3c224c9e70921a857ffcac32ccc7f4

### Platform

python

aravindone, 2023-05-29:
@bzbarsky-apple  @dhrishi please comment

dhrishi, 2023-05-29:
@aravindone What are you trying to do? The attributes of the descriptor cluster are not writable

aravindone, 2023-05-29:
im trying to input some dummy values for an attribute for that im trying some approaches , ESP 32 has all-clusters-app ..

I given these logs for example , i tried other clusters but it says unsupported attributes .

What will be the issue . how we can solve ?

dhrishi, 2023-05-29:
@aravindone Which attributes are you trying? If you try using the `write` command for a cluster, you will know which attributes are writable. You can then pass their ids to the `write-by-id` command along with appropriate values.

bzbarsky-apple, 2023-05-30:
What @dhrishi said.  Not all attribute are remote-writable, per spec.  Trying to write the non-writable ones will result in an UNSUPPORTED_WRITE error, as observed.

closed 2023-05-30 by bzbarsky-apple