project-chip/connectedhomeip #26932: [Feature] Setup Passcode supported in factory data provider

joe-kohler, 2023-05-30:
### Feature description

I make the factory data partition image with parameter -p PINCODE by tool generate_esp32_chip_factory_bin.py, but the ESP32FactoryDataProvider is without readout function
CHIP_ERROR GetSetupPasscode(uint32_t & setupPasscode) override { return CHIP_ERROR_NOT_IMPLEMENTED; }
src/platform/ESP32/ESP32FactoryDataProvider.h

the **populate_factory_data(args, spake2p_params)**:
 in generate_esp32_chip_factory_bin.py is without pin-code dump.

### Platform

esp32

### Platform Version(s)

4.4

shubhamdp, 2023-05-30:
As per some guidelines defined in `5.1.7. Generation of the Passcode`, verifier and passcodes shall be physically isolated. As passcode is not required to be stored in device, `ESP32FactoryDataProvider` do not store it.

If you want that feature, please feel free to extend `ESP32FactoryDataProvider` and implement the required methods set it using `SetCommissionableDataProvider()` API.

joe-kohler, 2023-05-30:
OK, Thanks

closed 2023-05-30 by joe-kohler

anuarrozman, 2023-09-01:
Hi @joe-kohler , did you manage to make and commission the factory data partition image with parameter -p PINCODE by tool generate_esp32_factory_bin.py ?