project-chip/connectedhomeip #26934: [Documentation]

patrickwang0821, 2023-05-30:
### Documentation issues

1.I would like to inquire about generating Matter QR Codes for a new device in the example. I have three Raspberry Pi 4 devices, with one being the Master and the other two being Slaves. However, I can only control one device, and the QR Codes generated for both devices are the same. I would like to know how to solve this issue. Is it currently only possible to have a one-to-one pairing in the Matter example?

### Platform

raspi

bzbarsky-apple, 2023-05-30:
> and the QR Codes generated for both devices are the same

How did you compile the applications in question?  How did you run them?

The example apps default to some default discriminator, but you can tell them to use a different one at either compile time or run time.

patrickwang0821, 2023-06-27:
> > and the QR Codes generated for both devices are the same
> 
> How did you compile the applications in question? How did you run them?
> 
> The example apps default to some default discriminator, but you can tell them to use a different one at either compile time or run time.

"I'm setting up my Matter development environment according to this link: https://community.arm.com/arm-community-blogs/b/internet-of-things-blog/posts/build-a-matter-home-automation-service-using-raspberry-pi-arm-virtual-hardware-and-python, but I haven't found a way to control more than two devices. Is there any place where I can find relevant information? When I replicate the slave end to another Raspberry Pi 4, the device information and QR code generated are the same, and the Master end can only control one device."

bzbarsky-apple, 2023-07-08:
I am not sure what the python bits use for passcode and discriminator....