project-chip/connectedhomeip #26947: Commissioning into a Thread network over Bluetooth LE "FAILURE"

BasemTarek, 2023-05-30:
Hi,

Simply, I'm using nRF5340-dk as a Matter light bulb and another one as a switch, and I'm using nRF52840 connected to a Raspberry Pi as a Thread Border Router to commission the nodes to the Thread network and Matter.

I followed this [documentation](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/matter/chip_tool_guide.html) to setup the OTBR, form the Thread network, and commission my nodes into the Thread network over BLE.
Also I'm using this [Snap packaging of chip-tool from the Matter project](https://github.com/canonical/chip-tool-snap)

When I run the following command pattern:
`sudo chip-tool pairing ble-thread 110 hex:<operational_dataset> <pin_code> <discriminator>`

I get this error:
`CHIP:BLE: No adapter available for new connection establishment`

Attached is the full logs file.
[ERORR.txt](https://github.com/project-chip/connectedhomeip/files/11600301/ERORR.txt)

Thanks,

bzbarsky-apple, 2023-05-30:
@BasemTarek From the log, this seems to be the relevant error:

```
[1685451869.515328][2542:2568] CHIP:DL: FAIL: Error getting object manager client: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: An AppArmor policy prevents this sender from sending this message to this recipient; type="method_call", sender=":1.53" (uid=0 pid=2542 comm="/snap/chip-tool/9/bin/chip-tool pairing ble-thread" label="snap.chip-tool.chip-tool (enforce)") interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" error name="(unset)" requested_reply="0" destination=":1.11" (uid=0 pid=917 comm="/usr/lib/bluetooth/bluetoothd " label="unconfined")
```

so chip-tool was not able o talk to bluetoothd, and then it can't commission things over bluetooth.

BasemTarek, 2023-05-30:
So, how can I solve this?

bzbarsky-apple, 2023-05-31:
@BasemTarek Check what AppArmor is and why it's denying access and how to fix that?