project-chip/connectedhomeip #26967: OTA testing with Apple/Alexa/Google[1.0]

SYESHASWINI, 2023-05-31:
### Reproduction steps

As the ota provider and requestor are apart, the ota process is done through the hubs. How can we test ota with apple/alexa/google platforms for matter devices.

### Bug prevalence

none

### GitHub hash of the SDK that was being used

none

### Platform

esp32

### Type

Test Improvement

bzbarsky-apple, 2023-05-31:
@SYESHASWINI Which part of the "OTA process" are you trying to test?  If you are trying to figure out whether apple/alexa/google will see published OTA updates in the DCL, that sounds like a question for those specific companies.  If not, what is "apple/alexa/google"-specific here?

SYESHASWINI, 2023-06-01:
@bzbarsky-apple we test ota using chip-tool, so is there a process for testing OTA with the Matter platforms.

bzbarsky-apple, 2023-06-01:
@SYESHASWINI I guess I don't understand what exactly you're trying to test.  Are you trying to test whether the device can accept an OTA offered by some provider?  Or are you trying to test whether the specific OTA providers that "apple/alexa/google" implement would offer updates for this particular device?

SYESHASWINI, 2023-06-02:
@bzbarsky-apple  I want to test how the providers apple/alexa/google can perform ota to matter devices.

bzbarsky-apple, 2023-06-02:
Then that's a question you should direct toward those providers, in their respective issue reporting systems...

closed 2023-06-02 by bzbarsky-apple