project-chip/connectedhomeip #26968: [1.0] Pairing using chip-tool over BLE fails with timeout

sigidagi, 2023-05-31:
### Reproduction steps

```
chip-tool pairing ble-thread 52 hex:some_hex 20202021 3840
```
Output:
```
[1685521936.164045][2310:2312] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 7E75AA5796B57A22)
[1685521936.176534][2310:2312] CHIP:CTL: Setting thread operational dataset from parameters
[1685521936.176572][2310:2312] CHIP:CTL: Setting attestation nonce to random value
[1685521936.176601][2310:2312] CHIP:CTL: Setting CSR nonce to random value
[1685521936.176622][2310:2312] CHIP:CTL: Commission called for node ID 0x0000000000000036
[1685521936.182306][2310:2311] CHIP:DL: TRACE: Bus acquired for name C-0906
[1685521936.201924][2310:2312] CHIP:DL: PlatformBlueZInit init success
[1685521936.218159][2310:2311] CHIP:BLE: BLE removing known devices.
[1685521936.220440][2310:2311] CHIP:BLE: BLE initiating scan.
[1685521936.534244][2310:2312] CHIP:DL: Long dispatch time: 358 ms, for event type 2
[1685521946.534529][2310:2312] CHIP:DIS: Closing all BLE connections
[1685521946.534764][2310:2312] CHIP:TOO: Pairing Failure: src/platform/Linux/bluez/ChipDeviceScanner.cpp:174: CHIP Error 0x00000032: Timeout
[1685521946.534909][2310:2312] CHIP:BLE: BLE scan error: src/platform/Linux/bluez/ChipDeviceScanner.cpp:174: CHIP Error 0x00000032: Timeout
[1685521946.543632][2310:2312] CHIP:BLE: Scan complete. No matching device found.
[1685521946.544258][2310:2310] CHIP:CTL: Shutting down the commissioner
```

bluetooth controller test:
1.
```
rock@rock5b:~/bin$ hciconfig -a
hci0:   Type: Primary  Bus: USB
        BD Address: 04:42:1A:5A:52:21  ACL MTU: 1021:6  SCO MTU: 255:12
        UP RUNNING
        RX bytes:1529 acl:0 sco:0 events:110 errors:0
        TX bytes:3853 acl:0 sco:0 commands:110 errors:0
        Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH HOLD SNIFF PARK
        Link mode: SLAVE ACCEPT
        Name: 'rock5b'
        Class: 0x0c0000
        Service Classes: Rendering, Capturing
        Device Class: Miscellaneous,
        HCI Version: 5.1 (0xa)  Revision: 0x999
        LMP Version: 5.1 (0xa)  Subversion: 0x646b
        Manufacturer: Realtek Semiconductor Corporation (93)
```
2.
```
rock@rock5b:~/bin$ bluetoothctl
Agent registered
[CHG] Controller 04:42:1A:5A:52:21 Pairable: yes
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# scan on
Discovery started
[CHG] Controller 04:42:1A:5A:52:21 Discovering: yes
[NEW] Device 30:BB:7D:12:A2:6C OnePlus 10T 5G
[NEW] Device 30:BB:7D:86:0C:7C OnePlus(Sigis)
[NEW] Device 4C:4F:EE:47:72:80 OnePlus8Due
[CHG] Device 30:BB:7D:12:A2:6C LegacyPairing: yes
[NEW] Device 72:8F:46:65:68:97 BV9900Pro
[CHG] Device 30:BB:7D:86:0C:7C LegacyPairing: yes
[CHG] Device 72:8F:46:65:68:97 RSSI: -64
[CHG] Device 72:8F:46:65:68:97 LegacyPairing: yes
[CHG] Device 30:BB:7D:86:0C:7C LegacyPairing: no
...
```
3.
```
[bluetooth]# connect 30:BB:7D:86:0C:7C
Attempting to connect to 30:BB:7D:86:0C:7C
[CHG] Device 30:BB:7D:86:0C:7C Connected: yes
[CHG] Device 30:BB:7D:86:0C:7C Modalias: bluetooth:v001Dp1200d1436
[CHG] Device 30:BB:7D:86:0C:7C UUIDs: 00001105-0000-1000-8000-00805f9b34fb
...
```

### Bug prevalence

often

### GitHub hash of the SDK that was being used

87994adc5c29cdabe366c321b1c62439ecf04710

### Platform

other

### Type

Platform Issue

sigidagi, 2023-06-01:
@bzbarsky-apple yes my device is advertising with discriminator 3840. But as I understand chip-tool should start scanning all bluetooth devices, like:
```
[1684401228.514820][314366:314371] CHIP:DL: PlatformBlueZInit init success
[1684401228.523306][314366:314369] CHIP:BLE: BLE removing known devices.
[1684401228.525451][314366:314369] CHIP:BLE: BLE initiating scan.
[1684401228.719445][314366:314371] CHIP:DL: Long dispatch time: 216 ms, for event type 2
[1684401228.808606][314366:314369] CHIP:BLE: Device 62:18:B4:D8:1B:E2 does not look like a CHIP device.
[1684401228.810710][314366:314369] CHIP:BLE: Device 18:46:44:41:85:E0 does not look like a CHIP device.
[1684401228.812762][314366:314369] CHIP:BLE: Device E9:B7:8A:3A:70:72 does not look like a CHIP device.
[1684401228.859622][314366:314369] CHIP:BLE: Device D4:9D:C0:A3:84:6F does not look like a CHIP device.
```
and that one which discriminator matches starts negotiation process.
But in my case   (see the issue above) non of the existing bluetooth devices (phones, TVs, etc.) has been scanned,

bzbarsky-apple, 2023-06-01:
@sigidagi Ah, in that case you might need to see what is going on with bluez in your case...

sigidagi, 2023-06-02:
@bzbarsky-apple Yes, in the description above you can see that I used `bluetoothctl` to scan and connect bluetooth devices manually. It looks for me that it is `chip-tool` issue.

chuckrui, 2023-08-16:
> @bzbarsky-apple Yes, in the description above you can see that I used `bluetoothctl` to scan and connect bluetooth devices manually. It looks for me that it is `chip-tool` issue.

Hi，I also encountered this problem. Have you solved it?

lion6420, 2023-09-21:
Hi @bzbarsky-apple 
We encountered this problem also.
We modified the `kConnectTimeout` from 10 second to 60 second in BLEManagerImpl. And then this problem is improved.
Is this modification reasonable?

bzbarsky-apple, 2023-09-21:
@lion6420 Seems pretty reasonable to me, but you should check with someone familiar with the Linux BLE bits.  Or just create a PR to that effect, better yet.

lion6420, 2023-09-22:
@bzbarsky-apple Thanks for your reply !
Is it better to make the BLEManager be able to be configured the different timeout value ?

bzbarsky-apple, 2023-09-22:
That might be nice too, but realistically controllers are not going to change this config much... We should select a value that actually works correctly with devices.