project-chip/connectedhomeip #26969: [BUG] Pairing using chip-tool over BLE fails with timeout

sigidagi, 2023-05-31:
### Reproduction steps

```
chip-tool pairing ble-thread 52 hex:some_hex 20202021 3840
```
Output:
```
[1685521936.164045][2310:2312] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 7E75AA5796B57A22)
[1685521936.176534][2310:2312] CHIP:CTL: Setting thread operational dataset from parameters
[1685521936.176572][2310:2312] CHIP:CTL: Setting attestation nonce to random value
[1685521936.176601][2310:2312] CHIP:CTL: Setting CSR nonce to random value
[1685521936.176622][2310:2312] CHIP:CTL: Commission called for node ID 0x0000000000000036
[1685521936.182306][2310:2311] CHIP:DL: TRACE: Bus acquired for name C-0906
[1685521936.201924][2310:2312] CHIP:DL: PlatformBlueZInit init success
[1685521936.218159][2310:2311] CHIP:BLE: BLE removing known devices.
[1685521936.220440][2310:2311] CHIP:BLE: BLE initiating scan.
[1685521936.534244][2310:2312] CHIP:DL: Long dispatch time: 358 ms, for event type 2
[1685521946.534529][2310:2312] CHIP:DIS: Closing all BLE connections
[1685521946.534764][2310:2312] CHIP:TOO: Pairing Failure: src/platform/Linux/bluez/ChipDeviceScanner.cpp:174: CHIP Error 0x00000032: Timeout
[1685521946.534909][2310:2312] CHIP:BLE: BLE scan error: src/platform/Linux/bluez/ChipDeviceScanner.cpp:174: CHIP Error 0x00000032: Timeout
[1685521946.543632][2310:2312] CHIP:BLE: Scan complete. No matching device found.
[1685521946.544258][2310:2310] CHIP:CTL: Shutting down the commissioner
```

bluetooth controller test:
1.
```
rock@rock5b:~/bin$ hciconfig -a
hci0:   Type: Primary  Bus: USB
        BD Address: 04:42:1A:5A:52:21  ACL MTU: 1021:6  SCO MTU: 255:12
        UP RUNNING
        RX bytes:1529 acl:0 sco:0 events:110 errors:0
        TX bytes:3853 acl:0 sco:0 commands:110 errors:0
        Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH HOLD SNIFF PARK
        Link mode: SLAVE ACCEPT
        Name: 'rock5b'
        Class: 0x0c0000
        Service Classes: Rendering, Capturing
        Device Class: Miscellaneous,
        HCI Version: 5.1 (0xa)  Revision: 0x999
        LMP Version: 5.1 (0xa)  Subversion: 0x646b
        Manufacturer: Realtek Semiconductor Corporation (93)
```
2.
```
rock@rock5b:~/bin$ bluetoothctl
Agent registered
[CHG] Controller 04:42:1A:5A:52:21 Pairable: yes
[bluetooth]# power on
Changing power on succeeded
[bluetooth]# scan on
Discovery started
[CHG] Controller 04:42:1A:5A:52:21 Discovering: yes
[NEW] Device 30:BB:7D:12:A2:6C OnePlus 10T 5G
[NEW] Device 30:BB:7D:86:0C:7C OnePlus(Sigis)
[NEW] Device 4C:4F:EE:47:72:80 OnePlus8Due
[CHG] Device 30:BB:7D:12:A2:6C LegacyPairing: yes
[NEW] Device 72:8F:46:65:68:97 BV9900Pro
[CHG] Device 30:BB:7D:86:0C:7C LegacyPairing: yes
[CHG] Device 72:8F:46:65:68:97 RSSI: -64
[CHG] Device 72:8F:46:65:68:97 LegacyPairing: yes
[CHG] Device 30:BB:7D:86:0C:7C LegacyPairing: no
...
```
3.
```
[bluetooth]# connect 30:BB:7D:86:0C:7C
Attempting to connect to 30:BB:7D:86:0C:7C
[CHG] Device 30:BB:7D:86:0C:7C Connected: yes
[CHG] Device 30:BB:7D:86:0C:7C Modalias: bluetooth:v001Dp1200d1436
[CHG] Device 30:BB:7D:86:0C:7C UUIDs: 00001105-0000-1000-8000-00805f9b34fb
...
```

### Bug prevalence

often

### GitHub hash of the SDK that was being used

87994adc5c29cdabe366c321b1c62439ecf04710

### Platform

other

bzbarsky-apple, 2023-05-31:
@sigidagi How is this different from https://github.com/project-chip/connectedhomeip/issues/26968 ?

sigidagi, 2023-06-01:
@bzbarsky-apple Oh, it seems duplication. I will close this one and I can follow https://github.com/project-chip/connectedhomeip/issues/26968

closed 2023-06-01 by sigidagi