project-chip/connectedhomeip #26970: [TC-LTIME-1.1, 1.2] Time Format Localization Cluster--Feature-map value mismatch.

Survensa, 2023-05-31:
**Problem:**
As per test plan [TC-LTIME-1.1, 1.2] FeatureMap attribute showing value mismatch error when executed, DUT should
responds with feature map value 1 but currently DUT responding feature map value has 0.

**Expected behavior:**- DUT should responds the feature map value as 1
**Actual behavior:**- DUT is responding the feature map value as 0

**Steps to reproduce:**- ./chip-tool timeformatlocalization read feature-map 1 0

**System configuration**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : Onnetwork
Commit-ID:  22de5a04f7c3f3d10bc06ad20bf50823c42cfbed
Dut application: all-clusters-app

**Test-Plan Reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/timeformatlocalization.adoc#tc-ltime-1-1-global-attributes-dut-as-client

**Spec Reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/LocalizationTimeFormat.adoc#TimeFormatLocalizationCluster

**Log:**
[LTIME.txt](https://github.com/project-chip/connectedhomeip/files/11611083/LTIME.txt)

bzbarsky-apple, 2023-05-31:
> Expected behavior:- DUT should responds the feature map value as 1

@Survensa Why?  Test plan says, for the expected behavior:

>  - bit 0 is set to 1 If CALFMT is true.

Is CALFMT true in this case?

Survensa, 2023-06-01:
@bzbarsky-apple CALFMT is set to true because the attributes "ActiveCalendarType" and "SupportedCalendarTypes" depending on the feature CALFMT - CalendarFormat is implemented so I believe that it should be listed in featuremap.

closed 2023-08-30 by mergify[bot]