project-chip/connectedhomeip #26976: [chiptool.py] Type mismatch on chiptool runner

ccruzagralopes, 2023-05-31:
### Reproduction steps / Feature

When running test cases that use a second commissioner (TH2), we get this error: `TypeError: ‘str’ object cannot be interpreted as an integer.`

### Platform

other

### Type

Manually tested with SDK

### Anything else?

Related issues:
- https://github.com/CHIP-Specifications/chip-certification-tool/issues/848
- https://github.com/CHIP-Specifications/chip-certification-tool/issues/842
- https://github.com/CHIP-Specifications/chip-certification-tool/issues/844

ccruzagralopes, 2023-05-31:
@mikaelhm please comment

mikaelhm, 2023-05-31:
`chiptool.py` is executed with this command:

```sh
./scripts/tests/yaml/chiptool.py tests Test_TC_CADMIN_1_16 --nodeId 0x67188cbf68c34be --delayInMs 250 --timeout 900 --continueOnFailure 1 --PICS /var/tmp/pics --trace_file "/logs/trace_log_2023-05-28_11.45.29_0x67188cbf68c34be_TEST_Test_TC_CADMIN_1_16.log" --trace_decode 1 --nodeId2 51966 --nodeId3 12586990 --endpoint 0 --waitAfterCommissioning 5000 --payload MT:-24J0AFN00KA0648G00
```

mikaelhm, 2023-05-31:
See related issue here: https://github.com/project-chip/connectedhomeip/issues/26303

cjandhyala, 2023-06-28:
@ccruzagralopes is this issue fixed with the PR #27008 ? Pls close the issue if the issue is fixed.

closed 2023-06-29 by ccruzagralopes