project-chip/connectedhomeip #26990: [BUG] binding unsuccessful

XiaoCongWu, 2023-06-01:
### Reproduction steps

Problem Description
Using the chip-tool tool to bind a switch(thread) light with Endpoint 3 of a bridge device, there is an issue with unsuccessful binding.
Prerequisites:
1.The switch(thread) has successfully joined the Matter network, with the node ID 6666 and button endpoint ID 1. This device can be controlled by chip-tool on a Raspberry Pi, and OTBR (OpenThread Border Router) is also running on the Raspberry Pi. (The application running on this device is the Matter EFR32 Light Switch Example.)
2.The bridge-app has successfully joined the Matter network, with the node ID 3333. This device can be controlled by chip-tool on the Raspberry Pi, but the application is not running on the Raspberry Pi.
Steps:
a. Write to the Access Control List.
b. Perform device binding.

### Bug prevalence

every day

### GitHub hash of the SDK that was being used

561d23d

### Platform

raspi

### Platform Version(s)

linux

### Anything else?

**1.Write to the Access Control List**
sudo ./chip-tool accesscontrol write acl '[{"fabricIndex":1, "privilege":5, "authMode":2, "subjects":[112233, 6666], "targets":null}]' 3333 3

**chip-tool print info:**
root@matter-hub:/home/ubuntu/work/connectedhomeip/out/standalone/chip-tool# sudo ./chip-tool accesscontrol write acl '[{"fabricIndex":1, "privilege":5, "authMode":2, "subjects":[112233, 6666], "targets":null}]' 3333 3
[1684388696.112393][26728:26728] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1684388696.127771][26728:26728] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1684388696.128283][26728:26728] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1684388696.128584][26728:26728] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1684388696.129479][26728:26728] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-EgY8uL)
[1684388696.130864][26728:26728] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1684388696.131001][26728:26728] CHIP:DL: NVS set: chip-counters/reboot-count = 69 (0x45)
[1684388696.132490][26728:26728] CHIP:DL: Got Ethernet interface: eth0
[1684388696.133530][26728:26728] CHIP:DL: Found the primary Ethernet interface:eth0
[1684388696.134671][26728:26728] CHIP:DL: Got WiFi interface: wlan0
[1684388696.134835][26728:26728] CHIP:DL: Failed to reset WiFi statistic counts
[1684388696.135018][26728:26728] CHIP:IN: UDP::Init bind&listen port=0
[1684388696.135278][26728:26728] CHIP:IN: UDP::Init bound to port=52299
[1684388696.135337][26728:26728] CHIP:IN: UDP::Init bind&listen port=0
[1684388696.135564][26728:26728] CHIP:IN: UDP::Init bound to port=38441
[1684388696.135621][26728:26728] CHIP:IN: BLEBase::Init - setting/overriding transport
[1684388696.135670][26728:26728] CHIP:IN: TransportMgr initialized
[1684388696.135741][26728:26728] CHIP:FP: Initializing FabricTable from persistent storage
[1684388696.136007][26728:26728] CHIP:TS: Last Known Good Time: 2023-05-17T15:55:27
[1684388696.139613][26728:26728] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x489331DFF68E882B, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1684388696.143920][26728:26728] CHIP:ZCL: Using ZAP configuration...
[1684388696.151076][26728:26728] CHIP:DL: MDNS failed to join multicast group on wpan0 for address type IPv4: ../../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:761: Inet Error 0x00000110: Address not found
[1684388696.151825][26728:26728] CHIP:CTL: System State Initialized...
[1684388696.151941][26728:26728] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1684388696.152074][26728:26728] CHIP:CTL: Setting attestation nonce to random value
[1684388696.152190][26728:26728] CHIP:CTL: Setting CSR nonce to random value
[1684388696.152324][26728:26728] CHIP:IN: UDP::Init bind&listen port=5550
[1684388696.152543][26728:26728] CHIP:IN: UDP::Init bound to port=5550
[1684388696.152604][26728:26728] CHIP:IN: UDP::Init bind&listen port=5550
[1684388696.152802][26728:26728] CHIP:IN: UDP::Init bound to port=5550
[1684388696.152857][26728:26728] CHIP:IN: TransportMgr initialized
[1684388696.153438][26728:26733] CHIP:DL: CHIP task running
[1684388696.154358][26728:26733] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1684388696.154516][26728:26733] CHIP:CTL: Setting attestation nonce to random value
[1684388696.154897][26728:26733] CHIP:CTL: Setting CSR nonce to random value
[1684388696.156942][26728:26733] CHIP:CTL: Generating NOC
[... 375 lines cut ...]
[1684388737.927765][2003:2003] CHIP:IN: Received a duplicate message with MessageCounter:29279840 on exchange 58366r
[1684388737.927948][2003:2003] CHIP:EM: >>> [E:58366r M:29279840] (U) Msg RX from 0:26C71BB36929B3BF [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
[1684388737.928061][2003:2003] CHIP:EM: Generating StandaloneAck via exchange: 58366r
[1684388737.928111][2003:2003] CHIP:EM: Forcing tx of solitary ack for duplicate MessageCounter:29279840 on exchange 58366r
[1684388737.928243][2003:2003] CHIP:EM: <<< [E:58366r M:180744978 (Ack:29279840)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1684388737.928529][2003:2003] CHIP:IN: (U) Sending msg 180744978 to IP address 'UDP:[fdde:d6d6:b14a:1:da95:14d5:5d48:fbda%eth0]:5540'
[1684388737.928806][2003:2003] CHIP:EM: Non-crit err ../third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:409: OS Error 0x02000065: Network is unreachable sending solitary ack for MessageCounter:29279840 on exchange 58366r
[1684388753.952685][2003:2003] CHIP:IN: Received a duplicate message with MessageCounter:29279840 on exchange 58366r
[1684388753.952801][2003:2003] CHIP:EM: >>> [E:58366r M:29279840] (U) Msg RX from 0:26C71BB36929B3BF [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
[1684388753.952893][2003:2003] CHIP:EM: Generating StandaloneAck via exchange: 58366r
[1684388753.952932][2003:2003] CHIP:EM: Forcing tx of solitary ack for duplicate MessageCounter:29279840 on exchange 58366r
[1684388753.953013][2003:2003] CHIP:EM: <<< [E:58366r M:180744979 (Ack:29279840)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1684388753.953191][2003:2003] CHIP:IN: (U) Sending msg 180744979 to IP address 'UDP:[fdde:d6d6:b14a:1:da95:14d5:5d48:fbda%eth0]:5540'
[1684388753.953372][2003:2003] CHIP:EM: Non-crit err ../third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:409: OS Error 0x02000065: Network is unreachable sending solitary ack for MessageCounter:29279840 on exchange 58366r

**switch RTT print info:**
00> <error > [SVR] Failed to establish connection to node 0x0000000000000D05
00> <error > [SVR] Failed to establish connection to node 0x0000000000000D05
00> <error > [SVR] Failed to establish connection to node 0x0000000000000D05
00> <error > [SVR] Failed to establish connection to node 0x0000000000000D05
00> <error > [SVR] Failed to establish connection to node 0x0000000000000D05
00> <error > [SVR] Failed to establish connection to node 0x0000000000000D05
00> <error > [SVR] Failed to establish connection to node 0x0000000000000D05
00> <error > [SVR] Failed to establish connection to node 0x0000000000000D05
00> <error > [EM] Failed to Send CHIP MessageCounter:29279840 on exchange 58366i sendCount: 4 max retries: 4

swap9391, 2023-06-01:
Hi  XiaoCongWu, 

_accesscontrol write acl '[{"fabricIndex":1, "privilege":5, "authMode":2, "subjects":[112233, 6666], "targets":null}]' 3333 3_

I am trying same command from mobile application using chiptool library, can you please tell me how to get  "privilege":5, "authMode":2, "subjects":[112233, 6666] values if my device is already commissioned ?

bzbarsky-apple, 2023-06-01:
>  sudo ./chip-tool accesscontrol write acl '[{"fabricIndex":1, "privilege":5, "authMode":2, "subjects":[112233, 6666], "targets":null}]' 3333 3

That command is obviously wrong: it's trying to write ACLs on endpoint 3 for some reason, and that fails, as expected, with UNSUPPORTED_CLUSTER in the log.

But the real problem is this, from the bridge-app:

> [1684388715.529065][2003:2003] CHIP:EM: <<< [E:58366r M:180744975 (Ack:29279840)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
> [1684388715.529581][2003:2003] [...]

So the bridge receives the Sigma1 from the switch, but when it tries to send back a Sigma2 that fails, and when it tries to send a StatusReport as a result that _also_ fails, because it does not know how to send a packet to that IP address.  That sounds like a network configuration problem with your network that you should fix.

closed 2023-06-01 by bzbarsky-apple