project-chip/connectedhomeip #26998: [BUG] Commission failed after PASE_Pake2

kacj1234j, 2023-06-01:
### Reproduction steps

Currently I'm doing the commission process but got StatusReport event sent from APP while status goes after PASE_Pake2 event.

The states PASE_Pake1 and PASE_Pake2 passed successfully, then suppossily I should received the next PASE_Pake3 (0x24) event, but I don't. 
I got StatusReport(0x40) and with the error message: "Received error (protocol code 2) during PASE process: 38", 
#define CHIP_ERROR_INVALID_PASE_PARAMETER CHIP_CORE_ERROR(0x38)

What may be the problem? Is it because that some information are not correct as scanned QRCode?
I used the QRcode which found at ESP website, and I modified the PIV, VID, Discriminator, Passcode are all same as website presents.

[https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#getting-the-repositories](https://docs.espressif.com/projects/esp-matter/en/main/esp32/developing.html#getting-the-repositories)

And in device certificate I chose example DAC for temporary certificate file.
SetDeviceAttestationCredentialProvider(Examples::GetExampleDACProvider());

Please help for some advice.
Thanks.

### Bug prevalence

During comission

### GitHub hash of the SDK that was being used

25423

### Platform

other

bzbarsky-apple, 2023-06-01:
> What may be the problem?

If it's failing at Pake3, that sounds like the PASE initiator is having an issue with the Pake2 it got?  Do you have logs from that PASE initiator @kacj1234j ?

kacj1234j, 2023-06-02:
> > What may be the problem?
> 
> If it's failing at Pake3, that sounds like the PASE initiator is having an issue with the Pake2 it got? Do you have logs from that PASE initiator @kacj1234j ?

Very thanks for the reply.

What is the PASE initiator means? 

I also trace on android APP log during comission, I saw an error message comes out "Failed to verify peer's MAC. This can happen when setup code is incorrect" when after recevied PAKE2 event.

Base on my knowns, the PASE process PBKDFParamRequest -> PKBDFParamResponse -> Pake1 -> Pake2, these steps will using Passcode (the configurable arguments for user) and do SPAKE+ algorithm to translate to a final key for generated a security channel. The algorithm should not be the root cause, so could the error is because of the Passcode?

Or did I just misunderstanding in these steps.

Thanks.

Regard.

dhrishi, 2023-06-02:
@kacj1234j If you are using esp-matter repo (as per the documentation you've shared), then please file issue [here](https://github.com/espressif/esp-matter/issues) and close this one.

>I used the QRcode which found at ESP website, and I modified the PIV, VID, Discriminator, Passcode are all same as website presents.

If you have modified these parameters, I assume you are using the mfg_tool. There would be a QR code that will be generated corresponding to these parameters and that should be used instead of the QR code that is only applicable for the default examples

kacj1234j, 2023-06-02:
> 
> If you have modified these parameters, I assume you are using the mfg_tool. There would be a QR code that will be generated corresponding to these parameters and that should be used instead of the QR code that is only applicable for the default examples

Very thanks for the reply,

I'm developing the matter repo directly, but just using ESP's QRcode to test the comission.
I have not using mfg_tool yet to generate the QRcode, but I had modified the param define that same as ESP's QRcode.

Vendor ID:                   65521
ProductID:                   32768
Long discriminator:     3840 (0xf00)
Passcode:                    20202021

I'll try the mfg_tool.

But on the other hand to your mention, is becuase the issue..
'an error message comes out "Failed to verify peer's MAC. This can happen when setup code is incorrect" when after recevied PAKE2 event on CHIP-Tool APP log'
.. will rely not only passcode but also other parameters ? (like PID VIP Discriminator?)

I also trace in to the "Failed to verify peer's MAC..." log located, I found the problem is becuase the previous steps "MacVerify" the "MAC" is not equal with "compute_mac" (CHIPCryptoPALmbedTLS.cpp)

But I have no idea what might cause the problem? I don't think the problem is on the algorithm, but which will be? 

Thanks.

Regard.

bzbarsky-apple, 2023-06-02:
> What is the PASE initiator means?

The thing that is trying to establish a PASE session.  In your case, the commissioner.

> "Failed to verify peer's MAC. This can happen when setup code is incorrect" 

Well, there is your issue.  What is the passcode the device is using?