project-chip/connectedhomeip #27027: [BUG] (Android) Commissioning to a custom fabric, FindOperational fails with "Invalid CASE parameter"

tnguyen-alarm, 2023-06-01:
### Reproduction steps

My phone: Samsung Galaxy S21 Ultra on Android 13

I have a Tapo P125M light switch that I'm attempting to commission to a custom fabric. I've got self-signed RCAC and use it to provide a NOC during commissioning. My CASE timeout is currently set to 30 seconds. This is what I'm seeing:

1. The process successfully completes PASE
2. Finds the right device with the compressed fabric ID and node ID
3. Attempts to establish CASE
4. Sends Sigma1. I noticed it never sends
5. Fails with "Going from commissioning step 'FindOperational' with lastErr = 'src/protocols/secure_channel/CASESession.cpp:1918: CHIP Error 0x00000054: Invalid CASE parameter'"

Any idea what might be going on here?

### Bug prevalence

Every time I try to commission the device

### GitHub hash of the SDK that was being used

8f66f4215bc0708efc8cc73bda80620e67d8955f

### Platform

android

### Platform Version(s)

1.1.0.1

### Anything else?

```
D  Performing next commissioning step 'FindOperational'
D  FindOrEstablishSession: PeerId = [2:0000000000000005]
D  FindOrEstablishSession: No existing OperationalSessionSetup instance found
D  OperationalSessionSetup[2:0000000000000005]: State change 1 --> 2
D  Resolving F7DD1B7222388850:0000000000000005 ...
D  resolve: Starting service resolution for 'F7DD1B7222388850-0000000000000005' type '_matter._tcp'
D  ICR moving to [AwaitingDe]
D  Checking node lookup status after 200 ms
D  Service discovery started. regType: _matter._tcp
D  onServiceFound: found target service name: F7DD1B7222388850-0000000000000005, type: _matter._tcp., hostAddresses: , port: 0, network: null, txtRecord: 
I  Discovery stopped: _matter._tcp
I  Resolved service 'F7DD1B7222388850-0000000000000005' to /fe80::32de:4bff:fef6:8cf, type : ._matter._tcp
D   ----- entry [0] : T 1
D   ----- entry [1] : SAI 300
D   ----- entry [2] : SII 5000
D  Node ID resolved for F7DD1B7222388850:0000000000000005
D  	Hostname: fe80::32de:4bff:
D  	IP Address #1: fe80::32de:4bff:fef6:8cf
D  	Port: 5540
D  	Mrp Interval idle: 5000 ms
D  	Mrp Interval active: 300 ms
D  	TCP Supported: 1
D  UDP:[fe80::32de:4bff:fef6:8cf]:5540: new best score: 3
D  Checking node lookup status after 301 ms
D  OperationalSessionSetup[2:0000000000000005]: Updating device address to UDP:[fe80::32de:4bff:fef6:8cf]:5540 while in state 2
D  OperationalSessionSetup[2:0000000000000005]: State change 2 --> 3
D  SecureSession[0x7aaeaaac88]: Allocated Type:2 LSID:30533
D  Initiating session on local FabricIndex 2 from 0x0000000000000005 -> 0x0000000000000005
D  KVS: Getting key f/2/g
D  KVS: Getting key f/2/k/0
D  KVS: Getting key f/2/s/0000000000000005
D  Key 'f/2/s/0000000000000005' not found in shared preferences
D  <<< [E:6234i S:0 M:122204406] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:30 (SecureChannel:CASE_Sigma1)
D  (U) Sending msg 122204406 to IP address 'UDP:[fe80::32de:4bff:fef6:8cf]:5540'
D  Sent Sigma1 msg
D  OperationalSessionSetup[2:0000000000000005]: State change 3 --> 4
D  >>> [E:6234i S:0 M:216774213 (Ack:122204406)] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
D  Found matching exchange: 6234i, Delegate: 0x7a9eab2d28
D  Rxd Ack; Removing MessageCounter:122204406 from Retrans Table on exchange 6234i
D  >>> [E:6234i S:0 M:216774214 (Ack:122204406)] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
D  Found matching exchange: 6234i, Delegate: 0x7a9eab2d28
D  CHIP MessageCounter:122204406 not in RetransTable on exchange 6234i
E  Received error (protocol code 2) during pairing process: src/protocols/secure_channel/CASESession.cpp:1918: CHIP Error 0x00000054: Invalid CASE parameter
D  SecureSession[0x7aaeaaac88]: Released - Type:2 LSID:30533
D  Device connection failed. Error src/protocols/secure_channel/CASESession.cpp:1918: CHIP Error 0x00000054: Invalid CASE parameter
D  Matter onCommissioningStatusUpdate nodeId: 5 stage: FindOperational errorCode: 84
D  Error on commissioning step 'FindOperational': 'src/protocols/secure_channel/CASESession.cpp:1918: CHIP Error 0x00000054: Invalid CASE parameter'
E  Failed to perform commissioning step 20
D  Going from commissioning step 'FindOperational' with lastErr = 'src/protocols/secure_channel/CASESession.cpp:1918: CHIP Error 0x00000054: Invalid CASE parameter' -> 'Cleanup'
D  Performing next commissioning step 'Cleanup' with completion status = 'src/protocols/secure_channel/CASESession.cpp:1918: CHIP Error 0x00000054: Invalid CASE parameter'
D  Matter onCommissioningStatusUpdate nodeId: 5 stage: Cleanup errorCode: 0
D  Successfully finished commissioning step 'Cleanup'
D  Matter onCommissioningComplete nodeId: 5 errorCode: 84
D  <<< [E:6234i S:0 M:122204407 (Ack:216774214)] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
D  (U) Sending msg 122204407 to IP address 'UDP:[fe80::32de:4bff:fef6:8cf%swlan0]:5540'
D  Flushed pending ack for MessageCounter:216774214 on exchange 6234i
```

bzbarsky-apple, 2023-06-02:
> Any idea what might be going on here?

For the other side to reject Sigma1?  Are the certs issued to the controller and the commissionee for the same fabric?  Are they using the same IPK?

tnguyen-alarm, 2023-06-02:
@bzbarsky-apple Thanks for the helpful questions.

They made me realize that the NOC that we issue to controller the is probably not the same NOC we pass to the device. It wasn't apparent in the device controller initialization API, but that makes sense because we don't have the device's public key yet. I guess that means that I'm commissioning my phone (the controller) to that fabric as well and it needs its own node ID?

And two your second question, there might be a bug in my code where it's using 2 different IPKs so I'll rectify that and test it out.

tnguyen-alarm, 2023-06-02:
Yep, everything is working now. It would be nice to update the API documentation for ChipDeviceController to be clearer about the operational certificate's purpose. I'll create a new issue for that.

closed 2023-06-02 by tnguyen-alarm