project-chip/connectedhomeip #27055: [BUG] ACE[1.2] TH2 does not receive a report from the DUT for subscription_noc

sumaky, 2023-06-03:
### Reproduction steps

**Setup Description**
1 | DUT | Commissionee
2 | TH1 | Commissioner - uses node ID N1
3 | TH2 | Administrator (client) - uses node ID N2, same fabric as N1
4 | TH3 | Commissioner - different fabric than TH1
N1-11223344
N2-4
**Steps Required to verify this issue**
1. execute the below mentioned command to put DUT into a commissionable state, Pls use equivalent command on the respective DUT
./chip-all-clusters-app
Once DUT reach the  commissionable state please  send below mentioned command on TH1.  
./chip-tool pairing onnetwork 1 20202021
Used Interactive mode to Verify this test case
/chip-tool interactive start
2. accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233,4], "targets": [{ "cluster": null, "endpoint": 0, "deviceType": null }]}]' 1 0
3. operationalcredentials subscribe nocs  1 30 1 0 --fabric-filtered false  --keepSubscriptions 1    --commissioner-nodeid 4
4. accesscontrol subscribe-event  access-control-entry-changed 1 30 1 0 --keepSubscriptions 1 --commissioner-nodeid 4
5. generalcommissioning subscribe breadcrumb 1 30 1 0 --keepSubscriptions 1 --commissioner-nodeid 4 
6. pairing open-commissioning-window 1 1 400 2000 3840
7. ON TH3 give the following command by using the Manual Pairing Code obtained from Pairing process from TH1 log after pairing:
./chip-tool pairing code 2 36049106779 --commissioner-name beta
**8.**TH2 does not receive a report from the DUT for subscription_noc****
**Issue:**
**Step 8: Expected Outcome**
TH2 receives a report from the DUT for subscription_noc. The AttributeReports list contains an entry with the AttributeData field present. That should contain a Path to the Node Operational Credentials cluster NOCs attribute.
Upto step 7 - all steps succeed

**For detailed log refer to the sheet** 
https://docs.google.com/spreadsheets/d/13zltSz50GRs_kidvobceEuIT8kAAHSK64ANyWcc3xQI/edit#gid=623879566

### Bug prevalence

NA

### GitHub hash of the SDK that was being used

NA

### Platform

other

sumaky, 2023-06-03:
@cecille

bzbarsky-apple, 2023-06-03:
The NOCs attribute has the "C" quality in the spec.  That means there are no subscription reports for it after the initial priming report, per spec.  Sounds like either the test plan or the verification steps need to be adjusted.  @cjandhyala

closed 2023-06-03 by bzbarsky-apple