project-chip/connectedhomeip #27060: [BUG] python: No module named esptool

lboue, 2023-06-05:
### Reproduction steps

Build bridge-app esp32
```
git clone https://github.com/project-chip/connectedhomeip.git
cd connectedhomeip
source scripts/bootstrap.sh
source scripts/activate.sh

cd ~/connectedhomeip/examples/bridge-app/esp32
gn gen out/debug
ninja -C out/debug
idf.py set-target esp32c3
idf.py build
```

### Bug prevalence

always

### GitHub hash of the SDK that was being used

275341ee0f21a9bf641f5dcd62c4f069f468c560

### Platform

esp32

### Platform Version(s)

N/A

### Anything else?

Do you know why esptool is not installed by bootstrap.sh script ?

I had to run that command:
 `pip install esptool`

Log:
[esptool-error.log](https://github.com/project-chip/connectedhomeip/files/11649448/esptool-error.log)

bzbarsky-apple, 2023-06-05:
@lboue I don't see you doing the export from the esp stuff anywhere in your commands above.  Are you doing it somewhere?

bzbarsky-apple, 2023-06-05:
And in particular, the steps above don't look very much like what https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/build_app_and_commission.md has listed.

shubhamdp, 2023-06-08:
@lboue What is the IDF version? In IDF v4.4.x `esptool.py` is part of repository, but from IDF v5.x and onwards it was moved out as an installable package. Since, docs still recommends using v4.4.x we have not added it to `scripts/setup/requirements.esp32.txt`.

This should have been noted in the docs, will fix it.

lboue, 2023-06-08:
Thanks. It will be fixed by PR #27141.

PR #27141 (2023-06-08): [ESP32] docs: install esptool for idf v5.x and above and disable "chip_build_tests" by default
Fixes #27060

Also, ESP32 examples were building examples with "chip_build_tests" set to "true", not needed, disabled them. May be builds would get a bit faster.
closed 2023-06-08 by lboue