project-chip/connectedhomeip #27061: [PICS TOOL v1.6.8] switch validation error for pics

maenkai, 2023-06-05:
### Reproduction steps / Feature

As shown in Figure, this is a mutually exclusive option that is validated using the pics tool only to be warned of an error
[image]

### Platform

other

### Platform Version(s)

v1.6.8

### Type

YAML tested

maenkai, 2023-06-06:
Yes, I have MS capabilities but I don't have LS capabilities
The tool's Web address is https://picstool.csa-iot.org/

maenkai, 2023-06-06:
> @maenkai This is almost certainly the wrong repository for this issue, since your problem is not in any of the Matter SDK code. @cjandhyala or @naveenkommareddi might know what the right repository is for the test harness bits.
> 
> That said, for Switch, you have to have either the LS feature or the MS feature set. Is one of those set in your case?

Yes, I have MS capabilities but I don't have LS capabilities
The tool's Web address is https://picstool.csa-iot.org/

bzbarsky-apple, 2023-06-06:
@maenkai Sounds like this should be filed in https://github.com/CHIP-Specifications/chip-certification-tool/issues

closed 2023-06-06 by bzbarsky-apple

maenkai, 2023-06-06:
> @maenkai Sounds like this should be filed in https://github.com/CHIP-Specifications/chip-certification-tool/issues

I cannot access the warehouse. What should I do

raju-apple, 2023-06-07:
Hi @maenkai please reach out to vtmgroupcsa@csa-iot.org to request getting added to the GitHub repo.

cjandhyala, 2023-06-07:
@maenkai Looks like you are using older version of PICS tool. if you are testing this for TE1 of fall release. pls use the appropriate PICS tool as mentioned in the TE1 slack channel.