project-chip/connectedhomeip #27071: [BUG] Reachable Attribute throws an 0x86 (UNSUPPORTED_ATTRIBUTE) Error.

Saravana-kr22, 2023-06-05:
### Reproduction steps

Bring up the DUT with all-clusters-app.
Pair the DUT using chip-tool through "onnetwork" transport.
Run the command `./chip-tool  basicinformation read reachable 1 0`

### Bug prevalence

Whenever i do this after the bfa05804ac3c224c9e70921a857ffcac32ccc7f4 commit in master branch

### GitHub hash of the SDK that was being used

bfa05804ac3c224c9e70921a857ffcac32ccc7f4 - Master

### Platform

raspi

### Platform Version(s)

RPI4 8GB RAM

### Anything else?

### Description:

In Basic-information cluster (BINFO), Reachable Attribute throws an 0x86 (UNSUPPORTED_ATTRIBUTE) Error.

**Expected behavior:-** DUT should respond with Success.

> [1658734602.363846][3240:3245] CHIP:DMG: }
[1658734602.364044][3240:3245] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0028 Attribute 0x0000_0011 DataVersion: 2381018651
[1658734602.364097][3240:3245] CHIP:TOO:   Reachable: TRUE
[1658734602.364223][3240:3245] CHIP:EM: Sending Standalone Ack for MessageCounter:267048399 on exchange 14115i

**Actual behavior:**- DUT is responding with UNSUPPORTED_ATTRIBUTE

> [1685962381.781839][12907:12909] CHIP:DMG: 					
[1685962381.781924][12907:12909] CHIP:DMG: 				StatusIB =
[1685962381.781999][12907:12909] CHIP:DMG: 				{
[1685962381.782074][12907:12909] CHIP:DMG: 					status = 0x86 (UNSUPPORTED_ATTRIBUTE),
[1685962381.782146][12907:12909] CHIP:DMG: 				},
[1685962381.782220][12907:12909] CHIP:DMG:

bzbarsky-apple, 2023-06-05:
The "reachable" attribute is typically only implemented for bridged devices implementing the Bridged Device Basic Information cluster.  You should typically _not_ expect it to be impemented in Basic Information cluster.  So I don't know where the "expected behavior" here is coming from.

closed 2023-06-05 by bzbarsky-apple

Trishia, 2023-10-17:
@bzbarsky-apple
But, TC-DD-3.18 Step 4a ~ 4b is verifying Reachable attribute of chip-all-clusters-app as TH1, TH2 even if chip-all-clusters-app is not Bridged Device.  How can I test **TC-DD-3.18** in this case ? 

Refer to this link: https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/suites/certification/Test_TC_DD_3_18.yaml#L123

bzbarsky-apple, 2023-10-17:
@Trishia Those specific verification steps are nonsense, sorry.  Ignoring for the moment what's going on with all-clusters-app, if that test were running against any sort of real device that device would almost certainly not have a Reachable attribute.

Trishia, 2023-10-18:
@bzbarsky-apple, I understand what are you saying about this issue. If so, TC-DD-3.18 Step 4a ~ 4b should be changed to other cluster attribute to read actions. Could you report this issue to proper certification staff ? I wish it would be fixed as soon as possible.

bzbarsky-apple, 2023-10-21:
@Trishia Please report issues in the cert bits at https://github.com/CHIP-Specifications/chip-test-scripts/issues

Trishia, 2023-10-23:
@bzbarsky-apple Unfortunately, I have no access permission to this repository of the link : https://github.com/CHIP-Specifications/chip-test-scripts/issues.

cjandhyala, 2023-10-25:
@Trishia you should have access to the test scripts repo now. Pls file an issue there.

cjandhyala, 2023-10-26:
@Trishia you should have access now , could you chk and raise the issue there ?