project-chip/connectedhomeip #27074: Nanoleaf Essential Smart LED bulb PAA attestation certificate for iMX93 [1.0]

Vardhman-Modi, 2023-06-05:
### Reproduction steps

I am using iMX93 as Open Thread Border Router OTBR and command line chip-tool itself only.

Now I want to test Certified Smart LED bulb (Nanoleaf Essential Smart LED). It showing me no PAA certification error during commissioning 

I need to do commission my custom thread border router from NXP to Nanoleaf Essential Smart LED bulb Matter A19 | E26 Smart Bulb (Each) https://nanoleaf.me/en-US/products/essentials/bulbs/?category=A19-E26&standard=matter&size=each.
I need PAA attestation certificate.

From where Can I find PAA attestation certificate for Nanoleaf Essential Smart LED bulb Matter A19 | E26 Smart Bulb.
Kindly help me on this.

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

master

### Platform

IMX8

### Type

Spec Compliance Issue

bzbarsky-apple, 2023-06-05:
The normal place to find PAA certificates is in the DCL (Distributed Compliance Ledger).

Vardhman-Modi, 2023-06-05:
Thank you for correct address to find PAA certificates that DCL. 
I have look on https://webui.dcl.csa-iot.org/pki, But did not able to find PAA certificates for NanoLeaf Essenetials.
Can you help me on this please.
Thanks,
Vardhman Modi

ndyck14, 2023-06-05:
Generally, you should ensure you have all PAA certificates in the DCL. Our PAIs trace up to StrongKey PAA.