project-chip/connectedhomeip #27083: Mode select XML does not match the spec

bzbarsky-apple, 2023-06-05:
The changes that landed in #26508 don't match the spec.  The spec has MfgCode in ModeTagStruct as nullable and required.  The XML that landed has it optional and non-nullable.

hicklin, 2023-06-05:
There has been some misalignment with which spec changes managed to get merged and which didn't. I had worked on the assumption that [this PR](https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/6989) will be merged. It seems that this is still open atm. That PR deprecates the `MfgCode` attribute. This is why I marked it as optional.

bzbarsky-apple, 2023-06-05:
Oh.  @hicklin I am 99% sure that PR will not be merged as-is...  It's got changes requested on it.  In fact, it's really not clear to me that the "old" Mode Select will have the changes that this PR made to it, or what subset of them it will have if it has them....

hicklin, 2023-06-16:
[This PR](https://github.com/project-chip/connectedhomeip/pull/27249/) has been merged following the ModeSelect/ModeBase spec change. I suspect we can now close this issue.

bzbarsky-apple, 2023-06-16:
Yes, this should be good now, I think.

closed 2023-06-16 by bzbarsky-apple