project-chip/connectedhomeip #27087: [BUG] DGSW-1.1 Test step 11 does not match test plan

jrhees-cae, 2023-06-05:
### Reproduction steps

Step 11 of the .yaml is conditional based on the following criteria:

PICS: DGSW.S.F00 && DGSW.S.Afffa

However, the EventList for the SoftwareDiagnostics cluster should only contain the event if the DGSW.S.E00 is TRUE in the PICS. Per the test plan, the criteria should read:

PICS: **DGSW.S.E00** && DGSW.S.Afffa

### Bug prevalence

Consistent

### GitHub hash of the SDK that was being used

1aca004d68640ae7df27adea09315308cf1913cd

### Platform

core

manjunath-grl, 2023-06-06:
@jrhees-cae fixed in PR [27011](https://github.com/project-chip/connectedhomeip/pull/27011) please verify it and close the ticket.

ccruzagralopes, 2023-06-29:
Closing this issue because it has been fixed. If this problem comes up again, please file a new issue

closed 2023-07-07 by woody-apple