project-chip/connectedhomeip #27091: [BUG] device out of power，application still found the device by mDNS

gen-zi, 2023-06-06:
### Reproduction steps

I use my own implementation of mDNS discovery which based on Android NsdManager to discovery matter devices, it actually work, but when i interrupt the power supply of device, it still can discovery the device, is this problem？

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

NA

### Platform

android

bzbarsky-apple, 2023-06-06:
DNS-SD records can get cached, like all DNS records.  So if you suddenly cut power to an advertiser without giving it a chance to send goodbye packets and invalidate the caches, queries will find the cached records until the TTL expires.

closed 2023-06-06 by bzbarsky-apple

gen-zi, 2023-06-25:
> DNS-SD records can get cached, like all DNS records. So if you suddenly cut power to an advertiser without giving it a chance to send goodbye packets and invalidate the caches, queries will find the cached records until the TTL expires.

can i invalidate the caches or disable caches in my app?

bzbarsky-apple, 2023-06-26:
@gen-zi The cache is on the DNS server.  The only way to invalidate the cache is to send the appropriate DNS advertisement packets to that server.

gen-zi, 2023-07-04:
ok，thank you for your reply

gen-zi, 2023-07-04:
i have one more question， i think the cache maybe effected to discover the service，but why i can also resolve the service，is the cache still working？