project-chip/connectedhomeip #27094: [BUG] [ACL-1.1] ACL's  event list does not contain records

doublemis1, 2023-06-06:
### Reproduction steps

1. Commission the DUT
2. Read event list attribute for ACL (./chip-tool accesscontrol read event-list 1 0

### Bug prevalence

always

### GitHub hash of the SDK that was being used

1aca004d68640ae7df27adea09315308cf1913cd

### Platform

nrf

### Platform Version(s)

nRF5340 DK door lock sample

### Anything else?

Found issue during Test Event 1 ACL-1.1
```
46.016883][4853:4855] CHIP:DMG: 				DataVersion = 0x2af71a96,
[1685950546.016974][4853:4855] CHIP:DMG: 				AttributePathIB =
[1685950546.017067][4853:4855] CHIP:DMG: 				{
[1685950546.017160][4853:4855] CHIP:DMG: 					Endpoint = 0x0,
[1685950546.017257][4853:4855] CHIP:DMG: 					Cluster = 0x1f,
[1685950546.017353][4853:4855] CHIP:DMG: 					Attribute = 0x0000_FFFA,
[1685950546.017443][4853:4855] CHIP:DMG: 				}
[1685950546.017539][4853:4855] CHIP:DMG: 					
[1685950546.017624][4853:4855] CHIP:DMG: 				Data = [
[1685950546.017696][4853:4855] CHIP:DMG: 						
[1685950546.017775][4853:4855] CHIP:DMG: 				],
[1685950546.017844][4853:4855] CHIP:DMG: 			},
[1685950546.017939][4853:4855] CHIP:DMG: 			
[1685950546.018010][4853:4855] CHIP:DMG: 		},
[1685950546.018089][4853:4855] CHIP:DMG: 		
[1685950546.018148][4853:4855] CHIP:DMG: 	],
[1685950546.018222][4853:4855] CHIP:DMG: 	
[1685950546.018288][4853:4855] CHIP:DMG: 	SuppressResponse = true, 
[1685950546.018348][4853:4855] CHIP:DMG: 	InteractionModelRevision = 1
[1685950546.018403][4853:4855] CHIP:DMG: }
[1685950546.018750][4853:4855] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_001F Attribute 0x0000_FFFA DataVersion: 720837270
[1685950546.018883][4853:4855] CHIP:TOO:   EventList: 0 entries
[1685950546.019158][4853:4855] CHIP:EM: <<< [E:24066i S:25993 M:235828743 (Ack:256056149)] (S) Msg TX to 1:0000000000000001 [A431] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1685950546.019249][4853:4855] CHIP:IN: (S) Sending msg 235828743 on secure session with LSID: 25993
```

doublemis1, 2023-06-06:
The same issue is in general diagnostics:
./chip-tool  generaldiagnostics read event-list 1 0

```
[1685950332.775728][4823:4825] CHIP:DMG: 					
[1685950332.775849][4823:4825] CHIP:DMG: 				Data = [
[1685950332.775924][4823:4825] CHIP:DMG: 						
[1685950332.776005][4823:4825] CHIP:DMG: 				],
[1685950332.776075][4823:4825] CHIP:DMG: 			},
[1685950332.776163][4823:4825] CHIP:DMG: 			
[1685950332.776224][4823:4825] CHIP:DMG: 		},
[1685950332.776296][4823:4825] CHIP:DMG: 		
[1685950332.776352][4823:4825] CHIP:DMG: 	],
[1685950332.776423][4823:4825] CHIP:DMG: 	
[1685950332.776480][4823:4825] CHIP:DMG: 	SuppressResponse = true, 
[1685950332.776538][4823:4825] CHIP:DMG: 	InteractionModelRevision = 1
[1685950332.776592][4823:4825] CHIP:DMG: }
[1685950332.776934][4823:4825] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0033 Attribute 0x0000_FFFA DataVersion: 981178976
[1685950332.777083][4823:4825] CHIP:TOO:   EventList: 0 entries
[1685950332.777347][4823:4825] CHIP:EM: <<< [E:58333i S:40950 M:187935125 (Ack:63156027)] (S) Msg TX to 1:0000000000000001 [A431] --- Type 0000:10 (SecureChannel:StandaloneAck)
```
Expected to have 1 entry with value 3.

bzbarsky-apple, 2023-06-06:
@doublemis1 I expect the issue is that the door lock sample does not enable these events in its ZAP config.

doublemis1, 2023-06-06:
event is not in the event list however the event is supported:
```
 ./chip-tool accesscontrol read event-list 1 0
[1686064339.222617][3039:3041] CHIP:DMG: ReportDataMessage =
[1686064339.222678][3039:3041] CHIP:DMG: {
[1686064339.222728][3039:3041] CHIP:DMG: 	AttributeReportIBs =
[1686064339.222802][3039:3041] CHIP:DMG: 	[
[1686064339.222859][3039:3041] CHIP:DMG: 		AttributeReportIB =
[1686064339.222938][3039:3041] CHIP:DMG: 		{
[1686064339.222999][3039:3041] CHIP:DMG: 			AttributeDataIB =
[1686064339.223083][3039:3041] CHIP:DMG: 			{
[1686064339.223162][3039:3041] CHIP:DMG: 				DataVersion = 0xf81cacfc,
[1686064339.223286][3039:3041] CHIP:DMG: 				AttributePathIB =
[1686064339.223386][3039:3041] CHIP:DMG: 				{
[1686064339.223470][3039:3041] CHIP:DMG: 					Endpoint = 0x0,
[1686064339.223569][3039:3041] CHIP:DMG: 					Cluster = 0x1f,
[1686064339.223666][3039:3041] CHIP:DMG: 					Attribute = 0x0000_FFFA,
[1686064339.223756][3039:3041] CHIP:DMG: 				}
[1686064339.223835][3039:3041] CHIP:DMG: 					
[1686064339.223922][3039:3041] CHIP:DMG: 				Data = [
[1686064339.223995][3039:3041] CHIP:DMG: 						
[1686064339.224073][3039:3041] CHIP:DMG: 				],
[1686064339.224140][3039:3041] CHIP:DMG: 			},
[1686064339.224216][3039:3041] CHIP:DMG: 			
[1686064339.224276][3039:3041] CHIP:DMG: 		},
[1686064339.224348][3039:3041] CHIP:DMG: 		
[1686064339.224402][3039:3041] CHIP:DMG: 	],
[1686064339.224473][3039:3041] CHIP:DMG: 	
[1686064339.224531][3039:3041] CHIP:DMG: 	SuppressResponse = true, 
[1686064339.224589][3039:3041] CHIP:DMG: 	InteractionModelRevision = 1
[1686064339.224643][3039:3041] CHIP:DMG: }
[1686064339.225015][3039:3041] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_001F Attribute 0x0000_FFFA DataVersion: 4162628860
[1686064339.227955][3039:3041] CHIP:TOO:   EventList: 0 entries
[1686064339.228321][3039:3041] CHIP:EM: <<< [E:49149i S:14444 M:59987642 (Ack:144924203)] (S) Msg TX to 1:0000000000000001 [BB02] --- Type 0000:10 (SecureChannel:StandaloneAck)

th_matter@thmatter:~/apps$ ./chip-tool accesscontrol read-event access-control-entry-changed 1 0

[1686064353.355047][3043:3045] CHIP:DMG: ReportDataMessage =
[1686064353.355107][3043:3045] CHIP:DMG: {
[1686064353.355156][3043:3045] CHIP:DMG: 	EventReportIBs =
[1686064353.355299][3043:3045] CHIP:DMG: 	[
[1686064353.355377][3043:3045] CHIP:DMG: 		EventReportIB =
[1686064353.355512][3043:3045] CHIP:DMG: 		{
[1686064353.355615][3043:3045] CHIP:DMG: 			EventDataIB =
[1686064353.355702][3043:3045] CHIP:DMG: 			{
[1686064353.355762][3043:3045] CHIP:DMG: 				EventPath =
[1686064353.355847][3043:3045] CHIP:DMG: 				{
[1686064353.355935][3043:3045] CHIP:DMG: 					Endpoint = 0x0,
[1686064353.356087][3043:3045] CHIP:DMG: 					Cluster = 0x1f,
[1686064353.356211][3043:3045] CHIP:DMG: 					Event = 0x0,
[1686064353.356280][3043:3045] CHIP:DMG: 				},
[1686064353.356383][3043:3045] CHIP:DMG: 				
[1686064353.356446][3043:3045] CHIP:DMG: 				EventNumber = 0x10002,
[1686064353.356589][3043:3045] CHIP:DMG: 				PriorityLevel = 0x1,
[1686064353.356680][3043:3045] CHIP:DMG: 				SystemTimestamp = 0x1ddd2,
[1686064353.356766][3043:3045] CHIP:DMG: 				EventData = 
[1686064353.356909][3043:3045] CHIP:DMG: 				{
[1686064353.357050][3043:3045] CHIP:DMG: 					0x1 = NULL
[1686064353.357121][3043:3045] CHIP:DMG: 					0x2 = 0, 
[1686064353.357228][3043:3045] CHIP:DMG: 					0x3 = 1, 
[1686064353.357294][3043:3045] CHIP:DMG: 					0x4 = 
[1686064353.357360][3043:3045] CHIP:DMG: 					{
[1686064353.357506][3043:3045] CHIP:DMG: 						0x1 = 5, 
[1686064353.357615][3043:3045] CHIP:DMG: 						0x2 = 2, 
[1686064353.357712][3043:3045] CHIP:DMG: 						0x3 = [
[1686064353.357841][3043:3045] CHIP:DMG: 								112233, 
[1686064353.358028][3043:3045] CHIP:DMG: 						],
[1686064353.358146][3043:3045] CHIP:DMG: 						0x4 = NULL
[1686064353.358224][3043:3045] CHIP:DMG: 						0xfe = 1, 
[1686064353.358295][3043:3045] CHIP:DMG: 					},
[... 14 lines cut ...]
[1686064353.359454][3043:3045] CHIP:TOO:   Timestamp: 122322
[1686064353.359655][3043:3045] CHIP:TOO:   AccessControlEntryChanged: {
[1686064353.359705][3043:3045] CHIP:TOO:     AdminNodeID: null
[1686064353.359765][3043:3045] CHIP:TOO:     AdminPasscodeID: 0
[1686064353.359811][3043:3045] CHIP:TOO:     ChangeType: 1
[1686064353.359877][3043:3045] CHIP:TOO:     LatestValue: {
[1686064353.359921][3043:3045] CHIP:TOO:       Privilege: 5
[1686064353.359964][3043:3045] CHIP:TOO:       AuthMode: 2
[1686064353.360017][3043:3045] CHIP:TOO:       Subjects: 1 entries
[1686064353.360113][3043:3045] CHIP:TOO:         [1]: 112233
[1686064353.360163][3043:3045] CHIP:TOO:       Targets: null
[1686064353.360207][3043:3045] CHIP:TOO:       FabricIndex: 1
[1686064353.360249][3043:3045] CHIP:TOO:      }
[1686064353.360291][3043:3045] CHIP:TOO:     FabricIndex: 1
[1686064353.360332][3043:3045] CHIP:TOO:    }
[1686064353.360624][3043:3045] CHIP:EM: <<< [E:20504i S:62726 M:186238820 (Ack:44010144)] (S) Msg TX to 1:0000000000000001 [BB02] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1686064353.360700][3043:3045] CHIP:IN: (S) Sending msg 186238820 on secure session with LSID: 62726
[1686064353.360993][3043:3045] CHIP:EM: Flushed pending ack for MessageCounter:44010144 on exchange 20504i
[1686064353.361321][3043:3043] CHIP:CTL: Shutting down the commissioner
[1686064353.361382][3043:3043] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1686064353.361430][3043:3043] CHIP:CTL: Shutting down the controller
[1686064353.361478][3043:3043] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1686064353.361521][3043:3043] CHIP:IN: SecureSession[0xffff9000ae60]: MarkForEviction Type:2 LSID:62726
[1686064353.361565][3043:3043] CHIP:SC: SecureSession[0xffff9000ae60, LSID:62726]: State change 'kActive' --> 'kPendingEviction'
```

bzbarsky-apple, 2023-06-06:
Yes, whether the event is supported for reads has nothing to do with the ZAP config.

cjandhyala, 2023-08-14:
since eventList is out of scope, closing the issue @doublemis1 feel free to re-open if you think this needs to be fixed.

closed 2023-08-14 by cjandhyala