project-chip/connectedhomeip #27099: [BUG] Hardcoded Software Diagnostic feature map in cluster implementation

doublemis1, 2023-06-06:
### Reproduction steps

Build sample on nRF52840 Door Lock with updated Software Diagnostic zap's feature map to 0.
The feature map value is overwritten by the hard coded cluster implementation.
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/software-diagnostics-server/software-diagnostics-server.cpp#L95

### Bug prevalence

always

### GitHub hash of the SDK that was being used

1aca004d68640ae7df27adea09315308cf1913cd

### Platform

nrf

### Platform Version(s)

nR

bzbarsky-apple, 2023-06-06:
@doublemis1 Well, does your device layer support watermarks?  If so, is there a reason you don't want your diagnostics cluster to support them?

bzbarsky-apple, 2023-06-06:
Sounds like you might want a compile-time flag to disable this behavior, then. But in general, people get feature maps wrong in ZAP all the time....

closed 2024-01-22 by doublemis1