project-chip/connectedhomeip #27110: Improving performance of the SDK

svigelj, 2023-06-06:
**Background:**
We are bridge-ing many devices (80, 100, 150+) via our Matter bridge device based on bridge-app example and using commit e2b9cadff4f15ac8fb3fa7c2144a945f229a6549 (May 30th 2023). A controller (eg. Apple Home) is making an incredible amount of attribute reads and it seems that the responses are not received fast enough for the controller to work correctly (the controller chokes and can’t be used). 

**Question:**
Let’s say that there are no hardware limitations and that our application specific response is instant. Can the Matter SDK be somehow configured to use more resources to improve performance?

We have been playing around with different settings in `src/lib/core/CHIPConfig.h`, `src/system/SystemConfig.h` and `src/include/platform/CHIPDeviceConfig.h` but with no success. Any tips would be much appreciated.

ste-ca, 2023-06-06:
Have you tried reducing the log level?

E.g. errors only: `chip::Logging::SetLogFilter(chip::Logging::LogCategory::kLogCategory_Error);`

bzbarsky-apple, 2023-06-06:
@svigelj Apple Home will generally do one read at a time, so you should not have problems servicing that on your end....

If you have situations where Apple Home is mis-behaving filing a feedback ticket with logs might be useful.  Please let me know the ticket id so I can follow up on it.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

svigelj, 2023-06-07:
@ste-ca yes, we have reduced the log level as well - which didn't help.

@bzbarsky-apple On our end we service all the requests without errors, the problem might be that Apple Home is expecting to go through all the reads in a shorter time than it does (it is making thousands of reads, especially for Reachable attribute). We will follow up with the feedback ticket as you have suggested shortly. Thanks!

svigelj, 2023-06-08:
@bzbarsky-apple the feedback ticket ID is **FB12268741**. We have included a detailed overview of the issue and how to reproduce it. Thank you for helping out on this issue :+1: .

bzbarsky-apple, 2023-06-13:
@svigelj Thank you!  Sorry for the slow response, but at this point we will hopefully have someone looking into this and figure out what's going on here.

woody-apple, 2023-06-23:
Closing this as this is being handled through feedback assistant.

closed 2023-06-23 by woody-apple