project-chip/connectedhomeip #27122: [BUG] TC-LVL-1.1: Getting incorrect value of feature map attribute of level control cluster

VaishaliAvhale, 2023-06-07:
### Reproduction steps

1. Commission a device
2. Check the below command
./chip-tool levelcontrol read feature-map 1 1
required value is 0 but getting 3
[lvl_1.1.log](https://github.com/project-chip/connectedhomeip/files/11676508/lvl_1.1.log)

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

1aca004d68640ae7df27adea09315308cf1913cd

### Platform

core

### Platform Version(s)

v1.2

bzbarsky-apple, 2023-06-07:
@VaishaliAvhale 3 means it has the On/Off and Lighting bits set, which is a completely reasonable thing to be doing if this is a dimmable light.  Why is "required value is 0" true here?

Looking at TC-LVL-1.1, the expected outcome for feature map is:

> Verify that the DUT response indicates that the FeatureMap attribute has bit 0 set to 1 if the DUT has a dependency on the On/Off cluster (i.e. LVL.S.F00(OO) is true), bit 1 set if the DUT supports lighting applications (i.e. LVL.S.F01(LT) is true)

What do the LVL.S.F* PICS for this device look like?

VaishaliAvhale, 2023-06-07:
It's an application side issue!!

VaishaliAvhale, 2023-06-07:
Sorry!!
It's an application-side issue!!

closed 2023-06-07 by VaishaliAvhale