project-chip/connectedhomeip #27137: [BUG] Identify in chip-tool vs Apple Home app

sam-gabbay, 2023-06-08:
### Reproduction steps

When calling the "./chip-tool identify identify 5 1234567895 1 --paa-trust-store-path /home/user_name/connectedhomeip/credentials/development/testing-certs" action, I receive a success and a function to the IdentifyStartHandler is called.

When calling "Identify" after pairing in the Apple Home app, nothing is called. Is this using a different identify method?

### Bug prevalence

N/A

### GitHub hash of the SDK that was being used

4083b32732b04aeec63b8f03ebd1e12ade78dd0f

### Platform

nrf

bzbarsky-apple, 2023-06-08:
@sam-gabbay What does the command payload the server receives look like when you do "Identify" after pairing in Apple Home?  If you can build with detail logging and `enable_im_pretty_print` set to true, that should log incoming payloads.

sam-gabbay, 2023-06-08:
@bzbarsky-apple I do not know what you did, but it started working. I think maybe an erase and wipe and a clean pair solved it.

P.S. You were something of a legend at Apple Park at WWDC. Kudos to you and all the work you do!

closed 2023-06-08 by sam-gabbay

bzbarsky-apple, 2023-06-08:
> P.S. You were something of a legend at Apple Park at WWDC. Kudos to you and all the work you do!

Thanks, and now I'm really curious what the stories were!