project-chip/connectedhomeip #27138: [TE#1][TC-G-2.2]The step2 of Pre-Conditions failed, Before adding the Group 0x0005 to check the Resource_Exhausted Condition, I got an General error: 0x01 (FAILURE)

Minan-Tuyasmart, 2023-06-08:
### Reproduction steps

**Test Case ID:**
TC-G-2.2

**Describe Issue:**
The step2 of Pre-Conditions failed, Before adding the Groups(MaxGroup+1) to check the Resource_Exhausted Condition, I got an General error: 0x01 (FAILURE)
`./chip-tool groupkeymanagement write group-key-map '[{"groupId": 13, "groupKeySetID": 1, "fabricIndex": 1} ]' 1 0`

For example， the "MaxGroupsPerFabric" value of my DUT is 4，after binding GroupId(0x0001 to 0x0004) with GroupKeySetID，I execute below command to bind GroupId(0x0005) with GroupKeySetID and then I got an General error: 0x01 (FAILURE)
`./chip-tool groupkeymanagement write group-key-map '[{"groupId": 5, "groupKeySetID": 1, "fabricIndex": 1} ]' 1 0`

**Steps that require review:**

I execute the below commands on the chip-tool side

1.
`./chip-tool groupkeymanagement read max-groups-per-fabric 1 0`
This step succeeded
2.
`./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": 1, "groupKeySecurityPolicy": 0, "epochKey0": "d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 2220000,"epochKey1": "d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 2220001,"epochKey2": "d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 2220002 }' 1 0`
This step succeeded
3.
`./chip-tool groupkeymanagement write group-key-map '[{"groupId": 1, "groupKeySetID": 1, "fabricIndex": 1},{"groupId":2, "groupKeySetID": 1, "fabricIndex": 1},{"groupId": 3, "groupKeySetID": 1, "fabricIndex": 1},{"groupId":4, "groupKeySetID": 1, "fabricIndex": 1} ]' 1 0`
This step succeeded
4.
` ./chip-tool groupkeymanagement write group-key-map '[{"groupId": 5, "groupKeySetID": 1, "fabricIndex": 1} ]' 1 0`
This step succeeded
5.
`./chip-tool groupkeymanagement write group-key-map '[{"groupId": 5, "groupKeySetID": 1, "fabricIndex": 1} ]' 1 0`
This step failed

**_actual outcome_**
[1686130382.225949][29543:29545] CHIP:DMG: WriteResponseMessage =
[1686130382.226009][29543:29545] CHIP:DMG: {
[1686130382.226058][29543:29545] CHIP:DMG: AttributeStatusIBs =
[1686130382.226142][29543:29545] CHIP:DMG: [
[1686130382.226201][29543:29545] CHIP:DMG: AttributeStatusIB =
[1686130382.226265][29543:29545] CHIP:DMG: {
[1686130382.226328][29543:29545] CHIP:DMG: AttributePathIB =
[1686130382.226476][29543:29545] CHIP:DMG: {
[1686130382.226559][29543:29545] CHIP:DMG: Endpoint = 0x0,
[1686130382.226655][29543:29545] CHIP:DMG: Cluster = 0x3f,
[1686130382.226749][29543:29545] CHIP:DMG: Attribute = 0x0000_0000,
[1686130382.226838][29543:29545] CHIP:DMG: }
[1686130382.226919][29543:29545] CHIP:DMG:
[1686130382.226992][29543:29545] CHIP:DMG: StatusIB =
[1686130382.227074][29543:29545] CHIP:DMG: {
[1686130382.227154][29543:29545] CHIP:DMG: status = 0x00 (SUCCESS),
[1686130382.227241][29543:29545] CHIP:DMG: },
[1686130382.227310][29543:29545] CHIP:DMG:
[1686130382.227371][29543:29545] CHIP:DMG: },
[1686130382.227459][29543:29545] CHIP:DMG:
[1686130382.227518][29543:29545] CHIP:DMG: AttributeStatusIB =
[1686130382.227581][29543:29545] CHIP:DMG: {
[1686130382.227643][29543:29545] CHIP:DMG: AttributePathIB =
[1686130382.227708][29543:29545] CHIP:DMG: {
[1686130382.227787][29543:29545] CHIP:DMG: Endpoint = 0x0,
[1686130382.227878][29543:29545] CHIP:DMG: Cluster = 0x3f,
[1686130382.227971][29543:29545] CHIP:DMG: Attribute = 0x0000_0000,
[1686130382.228062][29543:29545] CHIP:DMG: ListIndex = Null,
[1686130382.228148][29543:29545] CHIP:DMG: }
[1686130382.228236][29543:29545] CHIP:DMG:
[1686130382.228302][29543:29545] CHIP:DMG: StatusIB =
[1686130382.228377][29543:29545] CHIP:DMG: {
[1686130382.228455][29543:29545] CHIP:DMG: status = 0x01 (FAILURE),
[1686130382.228534][29543:29545] CHIP:DMG: },
[1686130382.228616][29543:29545] CHIP:DMG:
[1686130382.228675][29543:29545] CHIP:DMG: },
[1686130382.228750][29543:29545] CHIP:DMG:
[1686130382.228807][29543:29545] CHIP:DMG: ],
[1686130382.228887][29543:29545] CHIP:DMG:
[1686130382.228945][29543:29545] CHIP:DMG: InteractionModelRevision = 1
[1686130382.229003][29543:29545] CHIP:DMG: }

**_expect outcome_**
[1683799348.785746][39651:39653] CHIP:DMG: WriteResponseMessage =
[1683799348.785751][39651:39653] CHIP:DMG: {
[1683799348.785755][39651:39653] CHIP:DMG: AttributeStatusIBs =
[1683799348.785763][39651:39653] CHIP:DMG: [
[1683799348.785767][39651:39653] CHIP:DMG: AttributeStatusIB =
[1683799348.785771][39651:39653] CHIP:DMG: {
[1683799348.785775][39651:39653] CHIP:DMG: AttributePathIB =
[1683799348.785780][39651:39653] CHIP:DMG: {
[1683799348.785785][39651:39653] CHIP:DMG: Endpoint = 0x0,
[1683799348.785789][39651:39653] CHIP:DMG: Cluster = 0x3f,
[1683799348.785794][39651:39653] CHIP:DMG: Attribute = 0x0000_0000,
[1683799348.785798][39651:39653] CHIP:DMG: }
[1683799348.785804][39651:39653] CHIP:DMG:
[1683799348.785811][39651:39653] CHIP:DMG: StatusIB =
[1683799348.785816][39651:39653] CHIP:DMG: {
[1683799348.785821][39651:39653] CHIP:DMG: status = 0x00 (SUCCESS),
[1683799348.785825][39651:39653] CHIP:DMG: },
[1683799348.785831][39651:39653] CHIP:DMG:
[1683799348.785835][39651:39653] CHIP:DMG: },
[1683799348.785844][39651:39653] CHIP:DMG:
[1683799348.785848][39651:39653] CHIP:DMG: AttributeStatusIB =
[1683799348.785852][39651:39653] CHIP:DMG: {
[1683799348.785856][39651:39653] CHIP:DMG: AttributePathIB =
[1683799348.785860][39651:39653] CHIP:DMG: {
[1683799348.785865][39651:39653] CHIP:DMG: Endpoint = 0x0,
[1683799348.785869][39651:39653] CHIP:DMG: Cluster = 0x3f,
[1683799348.785874][39651:39653] CHIP:DMG: Attribute = 0x0000_0000,
[1683799348.785878][39651:39653] CHIP:DMG: ListIndex = Null,
[1683799348.785882][39651:39653] CHIP:DMG: }
[1683799348.785889][39651:39653] CHIP:DMG:
[1683799348.785894][39651:39653] CHIP:DMG: StatusIB =
[1683799348.785898][39651:39653] CHIP:DMG: {
[1683799348.785904][39651:39653] CHIP:DMG: status = 0x00 (SUCCESS),
[1683799348.785909][39651:39653] CHIP:DMG: },
[1683799348.785914][39651:39653] CHIP:DMG:
[1683799348.785918][39651:39653] CHIP:DMG: },
[1683799348.785924][39651:39653] CHIP:DMG:
[1683799348.785927][39651:39653] CHIP:DMG: ],
[1683799348.785936][39651:39653] CHIP:DMG:
[1683799348.785940][39651:39653] CHIP:DMG: InteractionModelRevision = 1
[1683799348.785943][39651:39653] CHIP:DMG: }

**chip-tool log**
[TC-G-2.2-chiptool.txt](https://github.com/CHIP-Specifications/chip-test-plans/files/11676583/TC-G-2.2-chiptool.txt)
[TC-G-2.2-DUT log.txt](https://github.com/CHIP-Specifications/chip-test-plans/files/11676587/TC-G-2.2-DUT.log.txt)

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

1aca004d68640ae7df27adea09315308cf1913cd

### Platform

efr32

q-thla, 2023-06-08:
Hi @Minan-Tuyasmart 
Why are you issuing the same command in step 4 and 5?

I also ran into failures with the preconditions.
On my end the add-group command failed.

steps:
- write group-key-map [of id 1 - 4]
- write group-key-map of id 5
- groups add-group 0x0001 grp1 1 1
I got an unsupported access error:
AddGroupResponse: {
  status: 126
  groupID: 1
 }
However, if I re-issue the write-group-key-map [of id 1-4]
the add-group command succeeds.

Minan-Tuyasmart, 2023-06-09:
@q-thla

Minan-Tuyasmart, 2023-06-09:
probably a copy paste error, I re-edited the issue and reuploaded the log of chiptool and DUT.

Minan-Tuyasmart, 2023-06-09:
@q-thla probably a copy paste error, I re-edited the issue and reuploaded the log of chiptool and DUT.

Minan-Tuyasmart, 2023-08-15:
@cjandhyala Yes, it's most likely an implementation issue on my device.

cjandhyala, 2023-08-25:
closing the issue per @Minan-Tuyasmart comments that it is product issue.

closed 2023-08-25 by cjandhyala