project-chip/connectedhomeip #27142: [TE1][TC-OO-1.1] Test Step 3 : Read the global attribute: FeatureMap. Test Failure : The test expectation "value == 0" is false

luojr9, 2023-06-08:
Reproduction steps
Pi TH run test case TC-OO-1.1. DUT is Midea Dishwasher.

Bug prevalence
xxx

GitHub hash of the SDK that was being used
xxx

Platform
asr

Platform Version(s)
xxx

Type
Test Event

Anything else?

Why the test expectation value is 0? My DUT replied value 1.

TH log
```
CHIPTOOL   | 2023-06-08 03:18:47.558047 | 		    [1mReceived[0m: [
CHIPTOOL   | 2023-06-08 03:18:47.559006 | 		                 {
CHIPTOOL   | 2023-06-08 03:18:47.559920 | 		                   "attribute": "FeatureMap",
CHIPTOOL   | 2023-06-08 03:18:47.560814 | 		                   "cluster": "On/Off",
CHIPTOOL   | 2023-06-08 03:18:47.561710 | 		                   "endpoint": 1,
CHIPTOOL   | 2023-06-08 03:18:47.562673 | 		                   "value": 1
CHIPTOOL   | 2023-06-08 03:18:47.563623 | 		                 }
CHIPTOOL   | 2023-06-08 03:18:47.564532 | 		               ]
CHIPTOOL   | 2023-06-08 03:18:47.565427 | 		***** Test Failure : The test expectation "value == 0" is false
DEBUG      | 2023-06-08 03:18:47.566907 | Found failure: The test expectation "value == 0" is false
WARNING    | 2023-06-08 03:18:47.567915 | Test Failure: The test expectation "value == 0" is false
CHIPTOOL   | 2023-06-08 03:18:47.568814 | [37m		4. R̶u̶n̶n̶i̶n̶g̶ G̶i̶v̶e̶n̶ ̶O̶O̶.̶S̶.̶F̶0̶0̶(̶L̶T̶)̶ ̶e̶n̶s̶u̶r̶e̶ ̶f̶e̶a̶t̶u̶r̶e̶m̶a̶p̶ ̶h̶a̶s̶ ̶t̶h̶e̶ ̶c̶o̶r̶r̶e̶c̶t̶ ̶b̶i̶t̶ ̶s̶e̶t[0m
CHIPTOOL   | 2023-06-08 03:18:47.569768 | 		***** Test Step 4 : Given OO.S.F00(LT) ensure featuremap has the correct bit set
INFO       | 2023-06-08 03:18:47.589595 | Test Step Completed [FAILED]: Read the global attribute: FeatureMap
INFO       | 2023-06-08 03:18:47.604065 | Executing Test Step: Given OO.S.F00(LT) ensure featuremap has the correct bit set
CHIPTOOL   | 2023-06-08 03:18:47.605107 | 		**** Skipping: OO.S.F00 == false
WARNING    | 2023-06-08 03:18:47.616732 | Test Step Not Applicable: Given OO.S.F00(LT) ensure featuremap has the correct bit set - Test step skipped: 		**** Skipping: OO.S.F00 == false
```
Detail log
[[TC-OO-1.1]_[jLUniBv1]_TH_log.log](https://github.com/project-chip/connectedhomeip/files/11684566/TC-OO-1.1._.jLUniBv1._TH_log.log)
[[TC-OO-1.1]_[jLUniBv1]_DUT_log.log](https://github.com/project-chip/connectedhomeip/files/11684567/TC-OO-1.1._.jLUniBv1._DUT_log.log)

ReneJosefsen, 2023-06-08:
If you DUT responds with a feature map of 1 for OnOff, it means that the lighting feature is enabled, which seems wrong? Is the DUT configuration correct?

ReneJosefsen, 2023-06-08:
What is your OO.S.F00(LT) & OO.S.F01(DF) PICS?

luojr9, 2023-06-08:
this?
```
        <features>
            <picsItem>
                <itemNumber>OO.S.F00</itemNumber>
                <feature>Level Control for Lighting</feature>
                <reference>Matter_App_Clusters_1.5.4.</reference>
                <status>O</status>
                <support>false</support>
            </picsItem>
        </features><!--Manually PICS write-->
```
[On-Off Cluster Test Plan.zip](https://github.com/project-chip/connectedhomeip/files/11687401/On-Off.Cluster.Test.Plan.zip)

luojr9, 2023-06-08:
This feature is true on module. Set it false and test PASS.

closed 2023-06-08 by luojr9