project-chip/connectedhomeip #27143: [BUG][TE #1][IDM-7.1]

joonhaengHeo, 2023-06-08:
### Reproduction steps

Run Note.

*Provision the device using chip tool on first controller(RD1)
 ./chip-tool pairing onnetwork 1 20202021
./chip-tool pairing open-commissioning-window 1 1 400 2000 3840 

*Provision the device using chip tool on 2nd  controller(RD2)
  ./chip-tool pairing code 2 36453420312(mannualcode) --commissioner-name beta
./chip-tool pairing open-commissioning-window 2 1 400 2000 3840 --commissioner-name beta

*Provision the device using chip tool on 3rd  controller(RD3)
./chip-tool pairing code 3 35603352859(mannualcode) --commissioner-name gamma
./chip-tool pairing open-commissioning-window 3 1 400 2000 3840 --commissioner-name gamma

*Provision the device using chip tool on 4th  controller(RD4)
 ./chip-tool pairing code 4 36375333175(mannualcode) --commissioner-name 4
 ./chip-tool pairing open-commissioning-window 4 1 400 2000 3840 --commissioner-name 4

*Provision the device using chip tool on 5th  controller(RD5)
  ./chip-tool pairing code 5 35358158796(mannualcode) --commissioner-name 5

Run  `./chip-tool interactive start`

Run Step 1
any subscribe-by-id '0x0028,0x0028,0x0028' '5,6,16' 100 1000 1 '0,0,0' --keepSubscriptions true

any subscribe-by-id '0x0028,0x0028,0x0028' '5,6,16' 100 1000 2 '0,0,0' **--commissioner-name beta** --keepSubscriptions true
   -> At this time, Commissionee is not found and ends with Timeout.

If the command specified in "Note" is executed after "interactive start", there is no problem in operation, but if interactive start is executed after commissioning, a problem occurs.

In interactive mode in chip-tool, the same "rootCA" certificate is returned regardless of commissioner-name.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

d4c7d47f4e6de1270ab5e246fe084ad16ec36e39

### Platform

raspi

bzbarsky-apple, 2023-06-08:
@joonhaengHeo Please attach logs from chip-tool and commissionnee, which include the exact commands executed.  When I just tried this locally, it works fine for me (though I only commissioned into alpha and beta fabrics).

> In interactive mode in chip-tool, the same "rootCA" certificate is returned regardless of commissioner-name.

Yes, this is true even in non-interactive mode.  The difference between the fabrics is the fabric id.

joonhaengHeo, 2023-06-09:
[IDM-7.1_Commissioner.txt](https://github.com/project-chip/connectedhomeip/files/11700861/IDM-7.1_Commissioner.txt)
[IDM-7.1_Commissionee.txt](https://github.com/project-chip/connectedhomeip/files/11700883/IDM-7.1_Commissionee.txt)

@bzbarsky-apple 
I attached logs from both sides.

For accurate analysis, both sides were processed with TH 2.9 images.
Commissioner: chip-tool
commissonee: all-cluster-app

I think fabric index 2 should be used instead of fabric index 6 being generated in the log below.

any subscribe-by-id '0x0028,0x0028,0x0028' '5,6,16' 100 1000 2 '0,0,0' --commissioner-name beta --keepSubscriptions true
[1686278118.051716][11125:11125] CHIP:TOO: Command: any subscribe-by-id 0x0028,0x0028,0x0028 5,6,16 100 1000 2 0,0,0 --commissioner-name beta --keepSubscriptions true
[1686278118.052647][11125:11127] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1686278118.052756][11125:11127] CHIP:CTL: Setting attestation nonce to random value
[1686278118.052959][11125:11127] CHIP:CTL: Setting CSR nonce to random value
[1686278118.054959][11125:11127] CHIP:CTL: Generating NOC
[1686278118.057955][11125:11127] CHIP:FP: Validating NOC chain
[1686278118.062671][11125:11127] CHIP:FP: NOC chain validation successful
[1686278118.063061][11125:11127] CHIP:FP: **Added new fabric at index: 0x6**
[1686278118.063127][11125:11127] CHIP:FP: Assigned compressed fabric ID: 0x82B44C33DD56B377, node ID: 0x000000000001B669
[1686278118.063195][11125:11127] CHIP:TS: Last Known Good Time: 2023-05-26T15:09:49
[1686278118.063252][11125:11127] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1686278118.063302][11125:11127] CHIP:TS: Retaining current Last Known Good Time
[1686278118.067841][11125:11127] CHIP:FP: Metadata for Fabric 0x6 persisted to storage.
[1686278118.075664][11125:11127] CHIP:TS: Committing Last Known Good Time to storage: 2023-05-26T15:09:49
[1686278118.085871][11125:11127] CHIP:CTL: Joined the fabric at index 6. Fabric ID is 0x0000000000000002 (Compressed Fabric ID: 82B44C33DD56B377)
[1686278118.213355][11125:11127] CHIP:TOO: Sending command to node 0x2
[1686278118.213787][11125:11127] CHIP:CSM: FindOrEstablishSession: PeerId = [6:0000000000000002]
[1686278118.213818][11125:11127] CHIP:CSM: FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1686278118.213854][11125:11127] CHIP:DIS: OperationalSessionSetup[6:0000000000000002]: State change 1 --> 2

joonhaengHeo, 2023-06-09:
@bzbarsky-apple 
It is confirmed that another root CA is generated when communicating in non-interactive mode, but in interactive mode, there is a problem of bringing the same root CA even if the name of the commissioner is changed.