project-chip/connectedhomeip #27145: ConcentrationMeasurement & ResourceMonitoring aliases provisional

ReneJosefsen, 2023-06-08:
A series of alias clusters of ConcentrationMeasurement & ResourceMonitoring is now marked as provisional in spec and needs to be mark in the SDK as well.

Spec PR (Already merged): https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/7130

bzbarsky-apple, 2023-06-08:
We need to decide what this means in practice.  Presumably we want to leave this in the XML so that chip-tool and YAML tests can work as needed?  That would also mean leaving it in XML for purposes of app-common codegen.

I suspect for similar reasons the python repl might want to keep this?  @tcarmelveilleux @cecille 

Past that, the Darwin API is marking these provisional in https://github.com/project-chip/connectedhomeip/pull/27152 so we don't ship APIs that are not ready.

What should Android and the Java APIs be doing here @andy31415?

andy31415, 2023-06-08:
Java has no direct "releases" to be tracked, so we can keep them being generated the same like we do for chip-tool (i.e. any controller API that one could write) or python.