project-chip/connectedhomeip #27165: Chip-tool has to be implemented for newly added clusters.

Survensa, 2023-06-09:
**Chip-tool command has to be implemented for below mentioned clusters:**
1. Dishwasher Mode Select Cluster.
2. Laundry Mode Select Cluster.
3. RVC Clean Mode Select Cluster.
4. RVC Operational State Cluster.
5. RVC Run Mode Select Cluster.
6. Refrigerator and Temperature Controlled Cabinet Cluster.
7. Washer Controls Cluster.

bzbarsky-apple, 2023-06-09:
@Survensa please _always_ list the SHA involved.  

* The mode select clusters you list have been supported in chip-tool since https://github.com/project-chip/connectedhomeip/pull/26508 merged.  This includes what I suspect you mean by "Refrigerator and Temperature Controlled Cabinet Cluster".
* RVC Operational State is being added in https://github.com/project-chip/connectedhomeip/pull/26989
* Washer Controls has been supported in chip-tool since https://github.com/project-chip/connectedhomeip/pull/26702 merged.

Survensa, 2023-06-13:
Verified with Master commit : 40fb7c246671e9d37ca8bb2cfbf37f81c1509334	

The only pending to be reflected in the chip-tool command is the RVC Operational State Cluster @bzbarsky-apple

bzbarsky-apple, 2023-06-13:
Yes, that's why that PR is linked to this issue.

closed 2023-06-15 by bzbarsky-apple