project-chip/connectedhomeip #27167: chip-tool requires Ethernet/WiFi

venlx, 2023-06-09:
I have a Raspberry Pi configured as a Thread Border Router and I have an Eve Smartplug (Thread over Matter).
I can only connect the Smartplug to my Thread Network via chip-tool when there is a connected Ethernet cable to the Raspberry Pi otherwise it fails.
Is this a restriction of the chip-tool or do I generally require an Ethernet/WiFi connection to be able to connect devices even though I'm only using Thread devices?

bzbarsky-apple, 2023-06-09:
> I can only connect the Smartplug to my Thread Network via chip-tool 

@venlx Which device is chip-tool running on?

bzbarsky-apple, 2023-06-09:
@venlx Please answer all the questions I asked.

venlx, 2023-06-13:
I used the following command:
sudo ./out/debug/standalone/chip-tool pairing code-thread 12 hex:0e080000000000010000000300001535060004001fffe00208088bf37a4a632ccc0708fdb466b36a9f5892051011223344556677889900aabbccddeeff030f4f70656e5468726561642d333439640102349d041090db5bc5927feeeaac3449f306170c050c0402a0f7f8 MT:-AM142O614LC.17NQ00 --paa-trust-store-path credentials/production/paa-root-certs/
(from the directory: /home/<username>/connectedhomeip)

[log.txt](https://github.com/project-chip/connectedhomeip/files/11735386/log.txt)

bzbarsky-apple, 2023-06-13:
OK, so the key part is this:

```
[1686662644.061056][25144:25147] CHIP:DIS: OperationalSessionSetup[1:000000000000000C]: operational discovery failed: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:114: CHIP Error 0x00000032: Timeout
```

Are you using minimal mdns or platform mdns for chip-tool?  Do you have avahi installed on this raspberry pi?  Does avahi see the operational advertisements in the condition when commissioning fails?

venlx, 2023-06-16:
How can I check if I'm using minimal mdns or platform mdns? I have avahi installed. I tried to use ```avahi-browse --all --resolve``` but it didn't see anything. Is that the right command to use or should I use something else?

bzbarsky-apple, 2023-06-16:
> How can I check if I'm using minimal mdns or platform mdns

How did you compile chip-tool?

As far as avahi goes... I am not sure what the right commands are to get this information, unfortunately.

venlx, 2023-06-23:
> How did you compile chip-tool?
I just used ```./gn_build.sh```

andy31415, 2023-06-23:
You could try to see what wireshard shows in terms of advertisemens for dnssd.

My guess is that since you have no wifi/ethernet interfaces, there are no packets going on the dnssd multicast addresses, so nothing will reply back.

You need chip-tool (which is not a thread-aware application) to be able to figure out dnssd data from thread devices (which goes over SRP I believe). This is generally done by the thread border router replying to DNSSD queries ... but it seems you are unable to query because you have no network interfaces to work with.

Maybe you could get it to work by setting up some local-only loopback networks or something. This is the first time I hear about such a setup though.

andy31415, 2023-06-23:
You can also try with platform dns configuration to see if avahi somehow manages to see these services (e.g. I am not sure how the thread border router does mdns ... if it uses avahi, maybe it just works because it would have avahi as a central shared registration and query source)

andy31415, 2023-06-23:
I would also recommend not using `gn_build.sh`, use `scripts/buid/build_examples.py`.

Examples for building chip-tool:

```
./scripts/build/build_examples.py --target linux-arm64-chip-tool
```

With platform mdns:

```
./scripts/build/build_examples.py --target linux-arm64-chip-tool-platform-mdns
```

bzbarsky-apple, 2023-06-23:
If you do use gn_build.sh, you can do `gn_build.sh 'chip_mdns="platform"'` to test.

caipiblack, 2025-02-26:
Hello,

This is just to say that I had the same issue:
- We have an IOT hub with Ethernet/WiFi, Thread border router and matter controller
- When we have no Ethernet/WiFi connection, the mDNS resolving was failing like in this issue
- After changing how the Matter stack is built to use the avahi implementation, it's now working fine!

If someone have this issue, just build the stack with `chip_mdns="platform"`.

Maybe this issue can be closed.