project-chip/connectedhomeip #27168: [FLDCONC] Attribute has to be implemented for Formaldehyde Concentration Measurement Cluster.

Survensa, 2023-06-09:
**Below mentioned attribute has to be implemented for Formaldehyde Concentration Measurement Cluster.**
1. MeasuredValue
2. MinMeasuredValue
3. MaxMeasuredValue
4. PeakMeasuredValue
5. PeakMeasuredValueWindow
6. AverageMeasuredValue
7. AverageMeasuredValueWindow
8. MeasurementUnit
9. MeasurementMedium

**Spec Reference :** https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/ConcentrationMeasurement.adoc#6-attributes

Survensa, 2023-06-21:
This issue Fixed in - https://github.com/project-chip/connectedhomeip/commit/f5c462133547c290f7eb09e5628ede162fb4a218

bzbarsky-apple, 2023-06-21:
@Survensa I don't see how https://github.com/project-chip/connectedhomeip/commit/f5c462133547c290f7eb09e5628ede162fb4a218 could possibly have fixed this.

Survensa, 2023-06-28:
@bzbarsky-apple It should be https://github.com/project-chip/connectedhomeip/commit/9354db4ce61d12c83d8de69529addb439607a86e, Please confirm so that I can close the issue

bzbarsky-apple, 2023-07-07:
Yep, that looks right, thank you!

closed 2023-07-07 by bzbarsky-apple