project-chip/connectedhomeip #27169: [NDOCONC] Attribute has to be implemented for Nitrogen Dioxide Concentration Measurement Cluster.

Survensa, 2023-06-09:
**Below mentioned attribute has to be implemented for Nitrogen Dioxide Concentration Measurement Cluster.**
1. PeakMeasuredValue
2. PeakMeasuredValueWindow
3. AverageMeasuredValue
4. AverageMeasuredValueWindow
5. LevelValue

**Spec Reference :** https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/ConcentrationMeasurement.adoc#6-attributes

bzbarsky-apple, 2023-06-09:
@Survensa I assume what you mean is you want these turned on in all-clusters-app?

Survensa, 2023-06-13:
> @Survensa I assume what you mean is you want these turned on in all-clusters-app?

Yes @bzbarsky-apple need this to be implemented in all-clusters-app

Survensa, 2023-06-21:
This issue Fixed in - f5c462133547c290f7eb09e5628ede162fb4a218

bzbarsky-apple, 2023-06-21:
I don't see how https://github.com/project-chip/connectedhomeip/commit/f5c462133547c290f7eb09e5628ede162fb4a218 could possibly have fixed this.

Survensa, 2023-06-22:
@bzbarsky-apple Could you check this log.
**Execution Log:** [TC-NDOCONC-1.1.txt](https://github.com/project-chip/connectedhomeip/files/11829649/TC-NDOCONC-1.1.txt), [TC-NDOCONC-2.1.txt](https://github.com/project-chip/connectedhomeip/files/11829650/TC-NDOCONC-2.1.txt)

Survensa, 2023-06-28:
@bzbarsky-apple It should be https://github.com/project-chip/connectedhomeip/commit/9354db4ce61d12c83d8de69529addb439607a86e, Please confirm so that I can close the issue

bzbarsky-apple, 2023-07-07:
Yep, that looks right, thank you!

closed 2023-07-07 by bzbarsky-apple