project-chip/connectedhomeip #27172: [BUG] TE1-V1.2: TC-G-2..2 & 2.3: Getting incorrect value of max-groups-per-fabric attribute of groupkeymanagement cluster

VaishaliAvhale, 2023-06-09:
### Reproduction steps

1. commission a device
2. followed steps from TC-G-2.2 below the command show incorrect output 
./chip-tool groupkeymanagement read max-groups-per-fabric 1 0

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

1aca004d68640ae7df27adea09315308cf1913cd

### Platform

esp32

### Platform Version(s)

v1.2

bzbarsky-apple, 2023-06-09:
@VaishaliAvhale What value are you getting?  What value are you expecting?  Why are you expecting the value you are expecting?

PSONALl, 2023-06-13:
the max-groups-per-fabric value should not be less than the minimum number of fabrics supported (i.e, 5) but we are getting value 4

bzbarsky-apple, 2023-06-13:
> the max-groups-per-fabric value should not be less than the minimum number of fabrics supported

Why not, @PSONALl ?  The number of groups ***per fabric*** is not related to the number of fabrics in any way.

PSONALl, 2023-06-14:
> Why not, @PSONALl ? The number of groups _**per fabric**_ is not related to the number of fabrics in any way.

right! My bad ! I misunderstood the test case

closed 2023-06-14 by VaishaliAvhale