project-chip/connectedhomeip #27174: SDK has to be implemented for newly added provisional clusters

Survensa, 2023-06-09:
**SDK has to implemented for below mentioned clusters:**

- [ ] Bromodichloromethane Concentration Measurement Cluster
- [ ] Bromoform Concentration Measurement Cluster
- [ ] Bromate Concentration Measurement Cluster
- [ ] Chloramines Concentration Measurement Cluster
- [ ] Chlorine Concentration Measurement Cluster
- [ ] Chlorodibromomethane Concentration Measurement Cluster
- [ ] Chloroform Concentration Measurement Cluster
- [ ] Copper Concentration Measurement Cluster
- [ ] Dissolved Oxygen Concentration Measurement Cluster
- [ ] Coli Concentration Measurement Cluster
- [ ] Ethylene Concentration Measurement Cluster
- [ ] Ethylene Oxide Concentration Measurement Cluster
- [ ] Fluoride Concentration Measurement Cluster
- [ ] Haloacetic Acids Concentration Measurement Cluster
- [ ] Hydrogen Concentration Measurement Cluster
- [ ] Hydrogen Sulfide Concentration Measurement Cluster
- [ ] Manganese Concentration Measurement Cluster
- [ ] Sodium Concentration Measurement Cluster
- [ ] Nitric Oxide Concentration Measurement Cluster
- [ ] Turbidity Concentration Measurement Cluster
- [ ] Oxygen Concentration Measurement Cluster
- [ ] Lead Concentration Measurement Cluster
- [ ] Sulfur Dioxide Concentration Measurement Cluster
- [ ] Sulfate Concentration Measurement Cluster
- [ ] Total Coliform Bacteria Concentration Measurement Cluster
- [ ] Total Trihalomethanes Concentration Measurement Cluster

bzbarsky-apple, 2023-06-09:
When you say "SDK has to be implemented", what exactly do you mean @Survensa ?

bzbarsky-apple, 2023-06-09:
That said, that looks like a list of the provisional concentration measurement clusters.  There are no plans to do any implementation work for those until there is a plan to have them not be provisional.

closed 2023-06-09 by bzbarsky-apple