project-chip/connectedhomeip #27175: [BUG] IDM-7.1 - Read the fabric scoped list with Fabric filtered set to false fails

PSONALl, 2023-06-09:
### Reproduction steps

1. Commission DUT using multiadmin with two admins
2. Follow steps in IDM -7.1
3. Commissioning with two admins is successful but when we read the fabric scoped list with Fabric filtered set to false using the below command from TH1 (Step 3 in IDM -7.1) gives unsupported access with TH1
     ./chip-tool operationalcredentials read fabrics 1 0 --fabric-filtered 0 

### Bug prevalence

EveryTime

### GitHub hash of the SDK that was being used

1aca004d68640ae7df27adea09315308cf1913cd

### Platform

core

### Platform Version(s)

v1.2

bzbarsky-apple, 2023-06-09:
@PSONALl Step 3 of IDM-7.1 does not involve a read, looking at https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#365-tc-idm-71-multi-fabric-subscription-test-cases-dut_server

Can you please show the complete set of chip-tool commands that you ran (starting with commissioning) to get that unsupported access error?

bzbarsky-apple, 2023-06-09:
Is it possible that you are talking about **IDM-8.1**?  If so, chances are that in step 2 the ACLs got messed up.  Again, knowing the exact commands used would be helpful in diagnosing what happened.

PSONALl, 2023-06-12:
@bzbarsky-apple yes my bad, editing the issue
Actual issue is I need to start new interactive session for each command: any subscribe-by-id '6,6,6' '0,0,0' 100 1000 5 '1,1,1' --commissioner-name 5 --keepSubscriptions true
Also I dont receive reports after changing subscribed attribute I see below logs on DUT attaching logs below
>  Failed to Send CHIP MessageCounter:265218867 on exchange 29652i sendCount: 4 max retries: 4

PSONALl, 2023-06-12:
Attaching logs:
[th-idm7.1.txt](https://github.com/project-chip/connectedhomeip/files/11723906/th-idm7.1.txt)

closed 2023-06-14 by PSONALl