project-chip/connectedhomeip #27176: [BUG]

tscheffl, 2023-06-09:
### Reproduction steps

I am trying to bootstrap matter on a Raspberry PI running Ubuntu 22.04 Server (32bit).
This has not gone as expected so far. I needed to manually install `rust` and `gn` because `activate.sh` was throwing some errors before, but now I truly am stuck with the following error message: 

```
ERROR at //BUILD.gn:80:18: Unable to load "/home/ubuntu/connectedhomeip/examples/common/pigweed/rpc_console/BUILD.gn".
    packages = [ "//examples/common/pigweed/rpc_console:chip_rpc_distribution" ]
                 ^------------------------------------------------------------
['gn', 'gen', '/home/ubuntu/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/ubuntu/connectedhomeip/third_party/pigweed/repo"']

Traceback (most recent call last):
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 364, in install_packages
    subprocess.check_call(
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gn', 'gen', '/home/ubuntu/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/ubuntu/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 878, in <module>
    sys.exit(main())
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 870, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 523, in setup
    result = step(spin)
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 683, in virtualenv
    if not virtualenv_setup.install(
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 399, in install
    install_packages(gn_target)
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 372, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['gn', 'gen', '/home/ubuntu/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/ubuntu/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.
```
The file/directory in question simply does not exist in the repo.
 

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

530349abc047f3d911216eac94c733bd6de4604d

### Platform

raspi

andy31415, 2023-06-09:
As an aside, https://github.com/project-chip/connectedhomeip/tree/master/examples/common/pigweed/rpc_console does exist currently and the `chip_rpc_distribution` target exists in there.
chip_rpc_distribution
I am not sure what is being run here - I need the actual commands as the logs do not actually explain it (I believe it is likely some wrong loading of packages or invalid chip_root that prevents the pacakge to be found)

tscheffl, 2023-06-09:
Thanks Andy, I see now that my examples folder is sparsely populated. I don't know why.
As regarding the `gn`, I was getting the following error before:

```
  Setting up CIPD package manager...done (9.2s)
  Setting up Python environment.....Traceback (most recent call last):
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 878, in <module>
    sys.exit(main())
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 870, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 523, in setup
    result = step(spin)
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 683, in virtualenv
    if not virtualenv_setup.install(
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 399, in install
    install_packages(gn_target)
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 364, in install_packages
    subprocess.check_call(
  File "/usr/lib/python3.10/subprocess.py", line 364, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.10/subprocess.py", line 345, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'gn'
```

andy31415, 2023-06-09:
Please provide the command that you are executing, not just the logs.

I assume you are running `activate.sh`? if so, you probably want `source ./scripts/bootstrap.sh` instead since you need a clean copy. If that fails, try really clean like `rm -rf .environment` first.

andy31415, 2023-06-09:
the sparsely populated directory also sounds odd. You probably need:

```
git clone https://github.com/project-chip/connectedhomeip.git
cd connectedhomeip
git submodule update -f --init --recursive 
```

instead of the full update you could also do 

```
./scripts/checkout_submodules.py --platform linux
```

for a less complete checkout

tscheffl, 2023-06-09:
I was doing the following: 
```
git clone --recurse-submodules --sparse https://github.com/project-chip/connectedhomeip.git
cd connectedhomeip
./scripts/checkout_submodules --shallow --platform linux
source scripts/activate.sh
```
Maybe the `./scripts/checkout_submodules --shallow --platform linux` was not such a good idea...

andy31415, 2023-06-09:
If you are having problems, please describe all your steps including *commands*, like what git clone, what cd, what other commands did you try.

sneha-raghunandanan, 2023-06-29:
I have the same error. I did the following
`git clone https://github.com/project-chip/connectedhomeip.git
cd connectedhomeip
git submodule update -f --init --recursive
cd 3rdParty
source ./scripts/bootstrap.sh`

This results in the following error
`BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (29.2s)
  Setting up Python environment.....[-]
ERROR at //build/config/linux/pkg_config.gni:116:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /home/sneha/connectedhomeip/3rdParty/.environment/gn_out/
Command: python3 /home/sneha/connectedhomeip/3rdParty/build/config/linux/pkg-config.py glib-2.0 gio-unix-2.0
Returned 1.
stderr:

Could not run pkg-config.

See //build/chip/linux/BUILD.gn:19:1: whence it was called.
pkg_config("glib") {
^-------------------
See //build/chip/linux/gdbus_library.gni:99:7: which caused the file to be included.
      glib_config,
      ^----------
['gn', 'gen', '/home/sneha/connectedhomeip/3rdParty/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/sneha/connectedhomeip/3rdParty/third_party/pigweed/repo"']

Traceback (most recent call last):
  File "/home/sneha/connectedhomeip/3rdParty/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 335, in install_packages
    subprocess.check_call(
  File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gn', 'gen', '/home/sneha/connectedhomeip/3rdParty/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/sneha/connectedhomeip/3rdParty/third_party/pigweed/repo"']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sneha/connectedhomeip/3rdParty/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 857, in <module>
    sys.exit(main())
  File "/home/sneha/connectedhomeip/3rdParty/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 849, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/sneha/connectedhomeip/3rdParty/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 501, in setup
    result = step(spin)
  File "/home/sneha/connectedhomeip/3rdParty/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 661, in virtualenv
    if not virtualenv_setup.install(
  File "/home/sneha/connectedhomeip/3rdParty/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 370, in install
    install_packages(gn_target)
  File "/home/sneha/connectedhomeip/3rdParty/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 343, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['gn', 'gen', '/home/sneha/connectedhomeip/3rdParty/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/sneha/connectedhomeip/3rdParty/third_party/pigweed/repo"']' returned non-zero exit status 1.`

Is there something else to try ? removing the environment and retrying also did not help!

andy31415, 2023-06-29:
I do not understand the steps ... what is up with the `cd 3rdParty`?

The build complains about `Command: python3 /home/sneha/connectedhomeip/3rdParty/build/config/linux/pkg-config.py glib-2.0 gio-unix-2.0`

Does ` /home/sneha/connectedhomeip/3rdParty/build/config/linux/pkg-config.py` exist? Path overall seems odd since why is this inside 3rdParty inside connectedhomeip? there should be no 3rdParty.

andy31415, 2023-06-29:
Closing this:
 - for @sneha-raghunandanan it seems that pkg-config was not installed
 - for @tscheffl I believe it is a bad checkout or no submodule checkout, since the folders exist in our repo.

closed 2023-06-29 by andy31415

tscheffl, 2023-07-03:
In my case the error started by a partial checkout, but was fundamentally caused by trying to compile the project on a 32-bit Ubuntu 22.04.
Some libraries are strictly 64-bit and there seems to be no 32-bit alternative, so the build ultimately fails.
I think there should be a disclaimer in the project not to try building it on a 32-bit Linux.

Regards, Thomas